Klipper3d/klipper #1121: Support for sam3xc controller

drphil3d, 2019-01-16:
how would one go about adding support for a SAM3X8C based controller?
I have the pin configuration file. I'm just not familiar with 32 bit controllers and don't quite know where to start.
https://github.com/synthetos/g2/blob/master/g2core/board/printrboardg2/printrboardG2v3-pinout.h

drphil3d, 2019-01-17:
I understand that, I don't think you understand what I'm doing.
https://vectormfg.net/klipper-multi-mcu/

Read the bottom of this page
https://github.com/KevinOConnor/klipper/blob/master/docs/Features.md

Klipper makes it possible to use the raspberry pi as a host device and the 8bit or 32bit Atmel MCU only for sending signals to the stepper motors and other outputs. The article I wrote above explains the benefits.

bshowket, 2019-01-17:
I am user of kivvy and I think it's a excellent idea. Specially for rpi2. My rpi takes about 10 minutes before I can use the web interface. Where kivvy will be ready instantly. Also print via octoprint causes pause for small move, so I have manually do sd card print most of the time. Kivvy is more responsive and with low power rpi and tft screen, it would be a great choice!

nikhil8333, 2019-01-17:
I clearly understand, but the issue I'm stating is 32 bit controllers are already expensive, and support multiple extruders or steppers like duet wifi which has support of additional 5 steppers other than 1X,1Y,2Z,2E. my point is instead of including expensive hardware we can use multiple 8 bit boards. yes there are boards like arduino due which are inexpensive being 32bit. but I feel why waste resources which are not necessary as 8bit controllers perform way better connected to rpi/klipper.

drphil3d, 2019-01-17:
There is still a limit to resolution and speed with an 8bit controller granted that limit is incredibly high. There is already 32bit arm support in klipper for 3sam8xe just not 8xc and I literally have piles of printrboad g2 controllers that I and many other people would really like to be able to use.

KevinOConnor, 2019-01-17:
It's not difficult to add support for the sam3x8c.  If you're willing to run tests on the hardware I can provide a test branch.  Let me know.

-Kevin

drphil3d, 2019-01-17:
Yes I would be very happy to

KevinOConnor, 2019-01-18:
I've uploaded a development branch with initial support for the sam3x8c mcu:

cd ~/klipper ; git fetch ; git checkout origin/work-sam3x8c-20190118 ; make menuconfig ; make ; sudo service klipper stop ; make flash ; sudo service klipper start

Note that this does not have USB support (only serial).  I'm not sure how the Printrboard G2 is wired so I don't know if that is important or not.

-Kevin

drphil3d, 2019-01-18:
when I used the 3x8c in the past it would show up on my computer as having two serial ports one a control port and the other a data port but you could use either one it didn't matter. I will try flashing it this afternoon.
looks like the g2core firmware exposes two usb serial ports. 
https://github.com/synthetos/g2/blob/master/g2core/board/printrboardg2/board_xio.h

I found the pin file
https://github.com/synthetos/g2/blob/master/g2core/board/printrboardg2/motate_pin_assignments.h

how to I convert these hardware pin numbers to something klipper can use?

Thank you so much by the way for your hard work!

KevinOConnor, 2019-01-20:
FYI, I was able to bring up USB support for the SAM3X series.  If you pull the latest version of the code you can now compile with USB support if you need that.

cd ~/klipper ; git fetch ; git checkout origin/work-sam3x8c-20190118 ; make menuconfig ; make ; sudo service klipper stop ; make flash ; sudo service klipper start

Klipper uses the micro-controller names for the pins - they have names like PA23 and PB7.  I don't know what pin naming convention those headers are using (though it looks like the header linked in the original post can be easily translated to the micro-controller  names).

-Kevin

drphil3d, 2019-01-21:
ok after a bit of strutting I got it flashed 🗡 
bossac-cli must be installed on linux

you have to activate the bootloader by opening a connection at 1200 baud
stty -F /dev/ttyACM0 1200 hup

to flash
pi@octopi:~/klipper/out $ bossac  --port=ttyACM0  -U true -e -w -v -i -b -R klipper.elf.hex

https://github.com/synthetos/g2/wiki/Flashing-g2core-with-Linux
pi@octopi:~/klipper/out $ stty -F /dev/ttyACM0 9600
pi@octopi:~/klipper/out $ bossac  --port=ttyACM0  -U true -e -w -v -i -b -R klipper.elf.hex
Erase flash
Write 61394 bytes to flash
[==============================] 100% (240/240 pages)
Verify 61394 bytes of flash
[==============================] 100% (240/240 pages)
Verify successful
Set boot flash true
Device       : ATSAM3X8
Chip ID      : 284e0a60
Version      : v1.1 Dec 15 2010 19:25:04
Address      : 524288
Pages        : 2048
Page Size    : 256 bytes
Total Size   : 512KB
Planes       : 2
Lock Regions : 32
Locked       : none
Security     : false
Boot Flash   : true

KevinOConnor, 2019-01-21:
The "make flash" command should work - the syntax would be something like: `make flash FLASH_DEVICE=/dev/ttyACM0`

Use a static_digital_output config section for the MS pins - see config/generic-rambo.cfg for an example and config/example-extras.cfg for a description of the config section.

-Kevin

drphil3d, 2019-01-21:
after flashing the board and power cycling It no longer shows up under dmesg or lsusb and the red light is on. the same light that comes on when you press reset button. So I think I flashed it with something it didn't like.

drphil3d, 2019-01-21:
I've tried pulling the code again, clean make file, tried with and without usb, etc. due boards wont flash with avrdude they have to have bossac. I can flash the hex file generated but it doesn't like it and just stays in booloader mode. I can flash tinyg firmware just fine.

KevinOConnor, 2019-01-22:
If you compile the micro-controller code for the SAM3X series (via make menuconfig) then make flash will invoke bossac.  If this isn't working, please run: `cd ~/klipper ; make clean ; make menuconfig ; make ; make flash FLASH_DEVICE=/dev/ttyACM0` and then cut-and-paste the entire terminal output to this github issue.

-Kevin

drphil3d, 2019-01-22:
pi@octopi:~/klipper $ cd ~/klipper ; make clean ; make menuconfig ; make ; make flash FLASH_DEVICE=/dev/ttyACM0
scripts/kconfig/mconf /home/pi/klipper/src/Kconfig

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

  Build Kconfig config file
  Creating symbolic link out/board
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/pwmcmds.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/avr/main.o
  Compiling out/src/avr/timer.o
  Compiling out/src/avr/gpio.o
  Compiling out/src/avr/adc.o
  Compiling out/src/avr/spi.o
  Compiling out/src/avr/i2c.o
  Compiling out/src/avr/hard_pwm.o
  Compiling out/src/avr/watchdog.o
  Compiling out/src/avr/usbserial.o
  Compiling out/src/generic/usb_cdc.o
  Building out/compile_time_request.o
Version: v0.7.0-97-gd942eb2-20190122_034448-octopi
  Linking out/klipper.elf
  Creating hex file out/klipper.elf.hex
  Flashing out/klipper.elf.hex to /dev/ttyACM0 via avrdude

Connecting to programmer: .avrdude: butterfly_recv(): programmer is not responding

avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
^Csrc/avr/Makefile:32: recipe for target 'flash' failed

KevinOConnor, 2019-01-22:
You've set the target to an AVR board.  You need to run "make menuconfig" and navigate the menu system there to select the SAM3X8C target.  Then run "make clean" then run "make flash FLASH_DEVICE=/dev/ttyACM0"

-Kevin

drphil3d, 2019-01-22:
YaY! that worked! the problem was I kept giving it a name when it asked to save the config

pi@octopi:~ $ cd ~/klipper ; make clean ; make menuconfig ; make ; make flash FLASH_DEVICE=/dev/ttyACM0
scripts/kconfig/mconf /home/pi/klipper/src/Kconfig

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

  Build Kconfig config file
  Creating symbolic link out/board
  Compiling out/src/sched.o
  Compiling out/src/command.o
  Compiling out/src/basecmd.o
  Compiling out/src/debugcmds.o
  Compiling out/src/gpiocmds.o
  Compiling out/src/stepper.o
  Compiling out/src/endstop.o
  Compiling out/src/adccmds.o
  Compiling out/src/spicmds.o
  Compiling out/src/thermocouple.o
  Compiling out/src/i2ccmds.o
  Compiling out/src/lcd_st7920.o
  Compiling out/src/lcd_hd44780.o
  Compiling out/src/buttons.o
  Compiling out/src/tmcuart.o
  Compiling out/src/atsam/main.o
  Compiling out/src/atsam/gpio.o
  Compiling out/src/atsam/i2c.o
  Compiling out/src/atsam/spi.o
  Compiling out/src/generic/crc16_ccitt.o
  Compiling out/src/generic/alloc.o
  Compiling out/src/generic/armcm_irq.o
  Compiling out/src/generic/timer_irq.o
  Compiling out/src/atsam/sam3_usb.o
  Compiling out/src/generic/usb_cdc.o
  Compiling out/src/atsam/adc.o
  Compiling out/src/atsam/timer.o
  Compiling out/src/../lib/sam3x/gcc/system_sam3xa.o
  Compiling out/src/../lib/sam3x/gcc/gcc/startup_sam3xa.o
  Building out/compile_time_request.o
Version: v0.7.0-97-gd942eb2-20190122_034937-octopi
  Linking out/klipper.elf
  Creating bin file out/klipper.bin
  Building bossac
make[1]: wx-config: Command not found
make[1]: wx-config: Command not found
CPP APPLET src/WordCopyArm.cpp
CPP COMMON src/Samba.cpp
CPP COMMON src/Flash.cpp
CPP COMMON src/D5xNvmFlash.cpp
CPP COMMON src/D2xNvmFlash.cpp
CPP COMMON src/EfcFlash.cpp
CPP COMMON src/EefcFlash.cpp
CPP COMMON src/Applet.cpp
CPP COMMON src/WordCopyApplet.cpp
CPP COMMON src/Flasher.cpp
CPP COMMON src/Device.cpp
CPP COMMON src/PosixSerialPort.cpp
CPP COMMON src/LinuxPortFactory.cpp
CPP BOSSAC src/bossac.cpp
CPP BOSSAC src/CmdOpts.cpp
LD bin/bossac
  Flashing out/klipper.bin lib/bossac/bin/bossac to /dev/ttyACM0 via bossac
Erase flash

Done in 0.030 seconds
Write 18556 bytes to flash (73 pages)
[==============================] 100% (73/73 pages)
Done in 0.657 seconds
Verify 18556 bytes of flash
[==============================] 100% (73/73 pages)
Verify successful
Done in 0.613 seconds
Set boot flash true

[27349.470435] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
[27350.679637] usb 1-1.3: USB disconnect, device number 34
[27351.176395] usb 1-1.3: new full-speed USB device number 35 using dwc_otg
[27351.526375] usb 1-1.3: device descriptor read/64, error -32
[27351.789406] usb 1-1.3: New USB device found, idVendor=2341, idProduct=abcd
[27351.789419] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[27351.789428] usb 1-1.3: Product: Klipper firmware
[27351.789436] usb 1-1.3: Manufacturer: Klipper
[27351.789445] usb 1-1.3: SerialNumber: 12345
[27351.790434] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device

lsusb
Bus 001 Device 035: ID 2341:abcd Arduino SA

drphil3d, 2019-01-22:
Awesome thank for your help! Now on to converting that pin file, the first version I made didn’t work sense the pin names aren’t arduino PIN format and the documentation on this mcu is sparse and hard to understand for someone who isn’t well versed in micro controllers.

KevinOConnor, 2019-01-22:
Great!  I've committed the sam3x8c changes to the master branch (commit eb7842c2).

If you run into a road block with the configuration then you can attach the klipper.log file (see https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md ) and we can take a look (though I'm not sure there is much we can do to help).  I suggest starting with config/generic-rambo.cfg and make sure to change all of the pins (everywhere there is a PA7/PH4/PJ2/etc pin in that file, be sure to change it to PA23/PB12/etc in the new file).  If you get a working config, it would be good to add it to the Klipper repo.

-Kevin

drphil3d, 2019-01-22:
this looks like what I need https://www.arduino.cc/en/Hacking/PinMappingSAM3X

drphil3d, 2019-01-22:
ok this is the last part I need some guidance on: setting the Vref voltage using pwm

the board has integrated DRV8825 drivers with vref mapped to PB17, PB18, PB19, PA2

this is just code I've pulled out of the g2 source to figure things out.
PWMOutputPin<vref_num> _vref;
 _vref{kNormal, frequency},
---
  void setPowerLevel(float new_pl) override {
        if (!_vref.isNull()) {
            _vref = new_pl;
        }
    };

    // sets default pwm freq for all motor vrefs (commented line below also sets HiZ)
    StepDirStepper(ioMode step_polarity, ioMode enable_polarity, const uint32_t frequency = 250000) :
        Stepper{},
        _step{step_polarity==IO_ACTIVE_LOW?kStartHigh:kStartLow},
        _enable{enable_polarity==IO_ACTIVE_LOW?kStartHigh:kStartLow},
        _vref{kNormal, frequency},
        _step_polarity{step_polarity},
        _enable_polarity{enable_polarity}
    {};

-----
// Stepper power management settings
#define Vcc         3.3                 // volts
#define MaxVref    2.25                 // max vref for driver circuit. Our ckt is 2.25 volts
#define POWER_LEVEL_SCALE_FACTOR ((MaxVref/Vcc)) // scale power level setting for voltage range

KevinOConnor, 2019-01-22:
It sounds like a pwm controlled voltage reference pin.  You'd want something similar to what's found in config/generic-mini-rambo.cfg .  Unfortunately, that requires hardware pwm support which has not been implemented for the sam3x.

-Kevin

drphil3d, 2019-01-23:
I see, makes sense why it rejects the code. darn 🤔 thats too bad, I will just have to wait until that happens.

KevinOConnor, 2019-01-23:
I took a look, and it doesn't seem simple to implement.

If you're feeling daring, you could try the following (be quick to pull the power if anything goes wrong):
```
[output_pin motor_x_pwm]
pin: PB17
pwm: True
scale: 2.25
cycle_time: 0.0001
```
Then at runtime manually send a `SET_OUTPUT_PIN PIN=motor_x_pwm VALUE=0.5` (where value is the desired amps) via the OctoPrint terminal to enable power to the motor.

-Kevin

drphil3d, 2019-01-23:
Awesome I'm getting somewhere with this. got the x stepper moving with out overheating and freaking out, it's super weak but it's moving.

[mcu]
serial: /dev/serial/by-id/usb-Klipper_Klipper_firmware_12345-if00
baud: 250000

[stepper_x]
step_pin: !PB15
dir_pin: PA16
enable_pin: !PB16
step_distance: .005
endstop_pin: !PB26
position_endstop: 0
position_max: 170

[stepper_y]
step_pin: PA21
dir_pin: PA26
enable_pin: !PA25
step_distance: .005
endstop_pin: PA11
position_endstop: 0
position_max: 220

[stepper_z]
step_pin: PA29
dir_pin: PA1
enable_pin: !PA0
step_distance: .005
endstop_pin: PA10
position_endstop: 0
position_max: 190

[extruder]
step_pin: PB14
dir_pin: PB23
enable_pin: !PB22
step_distance: .002
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PA5
sensor_type: EPCOS 100K B57560G104F
#sensor_pin: PA23
sensor_pin: PA4
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 250
 
#[heater_bed]
#heater_pin: PB24
#sensor_type: EPCOS 100K B57560G104F
#sensor_pin: PA4
#control: watermark
#max_temp: 110

[fan]
pin: PA6
#pin: PB25
#pin: PB27

#[heater_fan nozzle_cooling_fan]
#pin: B25

[printer]
kinematics: cartesian
max_velocity: 400
max_accel: 3000
max_z_velocity: 25
max_z_accel: 30

[output_pin motor_x_pwm]
pin: PB17
pwm: True
scale: 2.25
cycle_time: 0.0001

[static_digital_output 16xmicrostep_config]
pins:
	PA19, PB20, PA27, PB10

KevinOConnor, 2019-01-23:
I meant `SET_PIN PIN=motor_x_pwm VALUE=0.5`

-Kevin

drphil3d, 2019-01-23:
Yay! that works. I don't know how the software pwm is going to affect print
quality I think i can hear the pwm as the motor turns.

>

KevinOConnor, 2019-01-23:
FYI, hard pwm does not provide a benefit to heaters or fans, and Klipper does not use it for that purpose on any platform.

However, hard pwm is needed for setting of stepper voltage levels, and soft pwm would not work well in that role.  (It's purely about frequency of updates - .0001 is 10000 updates a second, whereas it looks like the original config uses 250000 updates per second.)  I would be curious, however, if you could verify the mapping of all the vref pins on the G2.  (Their choice of PA2, which is a timer pin, is odd considering the other pins appear to be PWMC pins.)

-Kevin

drphil3d, 2019-01-24:
oh interesting, that makes sense to me. I did find a file in the g2 firmware that looked like it was possibly remapping some of the pin configurations upon compiling. I will have to find it and show you, that could explain some of the strange pin mapping.

drphil3d, 2019-01-26:
Ok I have XYZ homing correctly and I can move all axis properly after setting pwm frequency
What doesn’t work yet:
heaters
Fans
Neopixel - super low priority

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

drphil3d, 2019-01-26:
well I can home xyz and move around just fine but the fans and heaters don't work

KevinOConnor, 2019-01-27:
Does the board have a global enable pin that must be set for the heaters / fans to be enabled?

-Kevin

drphil3d, 2019-01-27:
Not that I’m aware of. It has a uart output or it could be input but I kind
of doubt it that it globally disables anything. In the g2core pin file it
says safe signal rx tx that sends a signal of some kind from the MCU which
is confusing because there are two mcu. The printer has an esp8266 mounted
on the front of the machine with a touch screen to control it.
-- 

Philip Mally
3D Printing & Rapid Prototyping Services
530-205-3210
pmally@gmail.com

KevinOConnor, 2019-01-27:
I'm pretty sure the Klipper gpio code is working (it's tested on sam3x8e; it works on sam3x8c for steppers).  So, the likely explanation is an error in pin definitions or there is something board level (such as enable pins) that needs to be implemented.

Unfortunately, I don't have much advice on that - having a schematic for the board would help, but it doesn't look like there is one available.

-Kevin

drphil3d, 2019-01-27:
I can see about getting the board schematic. Thank you for all your help,
much appreciated.

KevinOConnor, 2019-02-08:
Any further updates on this?

-Kevin

drphil3d, 2019-02-08:
We have been working on it, slowly making progress.
https://github.com/MarlinFirmware/Marlin/issues/13020

KevinOConnor, 2019-02-08:
Interesting.  That's one strange heater/fan enable pin.  Out of curiosity, if the following is added to your printer.cfg file:
```
[output_pin heater_enable]
pin: PA7
pwm: True
cycle_time: .050
```
and then run `SET_PIN PIN=heater_enable VALUE=.004` at startup, does it allow you to use the fans/heaters?

-Kevin

drphil3d, 2019-02-10:
Check this out, have the heaters working in marlin2.0 thanks to Bob.

https://github.com/MarlinFirmware/Marlin/issues/13020#issuecomment-462086437

// Heaters / Fans
//
#define   HEATER_0_PIN              40 //                        G2 says PA5         DO_1
#define   HEATER_BED_PIN            41 //                        G2 says PB24        DO_11

#ifndef FAN_PIN
#define   FAN_PIN                   13 //           PB27 DO_4   Fan1A
#endif
#define   FAN1_PIN                  58 // was 60 // PA6  DO_3   Fan1B

#define FET_SAFETY_PIN              31 // PA7  must be pulsed low every 50 mS or FETs are turned off
#define FET_SAFETY_DELAY            50 // 50 mS delay between pulses
#define FET_SAFETY_INVERTED       true // true - negative going pulse of 2 uS

KevinOConnor, 2019-02-13:
FYI, I added support for hard pwm, at least for pins PB17, PB18, PB19 (commit 090cd930).

-Kevin

drphil3d, 2019-02-13:
Oh sweet thank you! So that means we should have an operational board including the FETs. Awesome, I just gotta test it.

drphil3d, 2019-02-19:
When I home the machine It hardly moves under its own power then gives an error
 Recv: !! Internal error on command:"G28"
[klippy.log](https://github.com/KevinOConnor/klipper/files/2878048/klippy.log)

KevinOConnor, 2019-02-19:
You'll need to set hard_pwm=True for the PB17/PB18/PB19 lines, and set the values in the config.  Look at config/generic-mini-rambo.cfg as an example.  Use a cycle_time=.000004 (which is 250Khz).  For now, I'd say just don't use the e motor (hardware pwm isn't available for the PA2 pin yet).

-Kevin

KevinOConnor, 2019-02-19:
FYI, the error is likely because you've set `homing_positive_dir=True` for stepper_y when that should not be set.  Remove that option from the config.

-Kevin

drphil3d, 2019-02-19:
still getting the same issue where it can hardly move

drphil3d, 2019-02-19:
omg hahaha silly me forgot to set the pwm value. Is there a way to have it pull this value so I don't have to enter it every time at runtime?
SET_PIN PIN=motor_x_pwm VALUE=0.5

KevinOConnor, 2019-02-19:
Yes - use hard_pwm and set a static_value in the config - see config/generic-mini-rambo.cfg for an example.  (Again, this wont yet work for the PA2 e motor pin.)

drphil3d, 2019-02-19:
Super Exciting news! That global enable for the mosfets works, fans and hotend work! so the only thing stopping this from printing is hard PWM on PA2 
SET_PIN PIN=heater_enable VALUE=.1

drphil3d, 2019-02-20:
ok I'm running into a strange issue now where I set the hotend to heatup and as it climbs to around 100c it starts slowing down and then shuts down around 105c even though the hotend is has to be over 200c because I can easily push filament through. I believe the issue may be due  to the thermistor voltage being 3.3v instead of 5v. I just get an error if I try to set the thermistor voltage to anything less.
I tried creating thermistor lookup table but at around 110c it stalls.

[klippy.log.2019-02-1.txt](https://github.com/KevinOConnor/klipper/files/2883349/klippy.log.2019-02-1.txt)

adc_voltage: 3.3

drphil3d, 2019-02-21:
It's pulled up to 3.3V but the pullup resistance is 9.4K a
This is a thermistor table made for marlin specific for the g2
[G2_therm table.txt](https://github.com/KevinOConnor/klipper/files/2888550/G2_therm.table.txt)
nd the signal is divided by 2

KevinOConnor, 2019-02-22:
Voltage isn't used for thermistors.  (The resistance of the thermistor is measured relative to the resistance of the pullup, and therefore the voltage drops out of the equations.)

It sounds like you have three temperature and resistance pairs for the thermistor - you can create a custom [thermistor] section in the config file and add the three pairs - see the config/example-extras.cfg file for the details.

-Kevin

drphil3d, 2019-03-22:
Ok, I have a thermistor table that works in marlin. I tested it with a thermocouple. How do I convert this to something Klipper can use?
[thermistor_80.h.txt](https://github.com/KevinOConnor/klipper/files/2994768/thermistor_80.h.txt)

KevinOConnor, 2019-03-22:
You'd add a custom thermistor setting with:
```
[thermistor B57540G0]
t1: 25
r1: 100000
t2: 150
r2: 16419
t3: 300
r3: 10716
```

Then you can use the `B57540G0` as your sensor_type in your heater section.  (The temp/resistance pairs are from a spec I found online for that thermistor type - if you have temp/resistance pairs that you've measured directly, then you can use those values instead.)

-Kevin

KevinOConnor, 2019-03-22:
Actually, the example above is incorrect.  If you have a B57540G0, then it directly matches the existing `EPCOS 100K B57560G104F` setting.  Do you know what type of thermistor you actually have?  If you don't what info do you have from your thermocouple?

-Kevin

drphil3d, 2019-03-25:
From the G2 Firmware R25 = 100 kOhm,Epcos B57540G0104G000

If I use the EPCOS 100K B57560G104F the tempature never reads past 100c yet my multimeter thermocouple is reading 235c
Thermistor vs Thermocouple

Using this custom table pull up resistor set to 4700
I'm reading 
Thermistor vs Thermocouple
Set 100C  Reading: 100c / 50c
Set 200C Reading 200c / 73c
Set 250C Reading 213c / 74c

I set the pull up resistor value to 9400 which resulted in 
200 / 200 / 108

[thermistor B57540G0]
temperature1: 25
resistance1: 100000
temperature2: 150
resistance2: 16419
temperature3: 300
resistance3: 10716

so confused

drphil3d, 2019-03-25:
ok using two multimeters and a touch lighter I was able to read the resistance of the thermistor and the thermocouple tempature
25c 100k
30c 7500k
82c 10.4k
150c 1.4k

drphil3d, 2019-03-25:
This is the closest I've come to reading correct temp

[thermistor B57540G0CUS4]
temperature1: 25
resistance1: 88500
temperature2: 150
resistance2: 7019
temperature3: 300
resistance3: 1316

Pull Up Resistor set to 9400

drphil3d, 2019-03-26:
I'm looking at the G2 source code to try and figure out what they did. 
https://github.com/synthetos/g2/blob/master/g2core/temperature.cpp

 My interpretation of the Formulas they use

Voltage = RAW ADC*3.3v       //convert the 10 bit ADC value to a voltage
Resistance = 4700*3.3/3.3 - 4700       // resistance of thermistor

[thermistor G2]
temperature1: 20
resistance1: 140000
temperature2: 195
resistance2:  593.0
temperature3: 255
resistance3: 189

```
float v = (float)raw_adc_value * kSystemVoltage / (adc_pin.getTop()); // convert the 10 bit ADC value to a voltage
float r = ((pullup_resistance * v) / (kSystemVoltage - v)) - inline_resistance;   // resistance of thermistor

        float lnr = log(r);
        float Tinv = c1 + (c2*lnr) + (c3*pow(lnr,3));
        return (1/Tinv) - 273.15; // final temperature
```

Thermistor Table Values

```
// Extruder 1
Thermistor<kADC1_PinNumber> thermistor1 {
    /*T1:*/     20.0, /*T2:*/  195.0, /*T3:*/ 255.0,
    /*R1:*/ 140000.0, /*R2:*/  593.0, /*R3:*/ 189.0, /*pullup_resistance:*/ 4700, /*inline_resistance:*/ 4700
    };
```

KevinOConnor, 2019-03-27:
Yeah, from the `Thermistor<kADC1_PinNumber> thermistor1` settings, your `[thermistor G2]` block looks correct.  Does it work okay?

-Kevin

drphil3d, 2019-03-30:
Well I have some good news I can accurately read temperatures under 100c but as soon as I go higher than that it goes wonky.
Set  Read Actual
80     80      80
100  102    124
200  190    200

[G2 Reading up to 100c.cfg.txt](https://github.com/KevinOConnor/klipper/files/3025161/G2.Reading.up.to.100c.cfg.txt)

[thermistor G2]
temperature1: 20
resistance1: 140000
temperature2: 195
resistance2: 593
temperature3: 255
resistance3: 189

[51 more replies]

closed 2019-06-21 by drphil3d