Klipper3d/klipper #1124: Fresh install of Klipper for a CR-10s

Ozkovacs, 2019-01-17:
Hi there.
I had my CR-10s for a couple weeks now but I've always used the Marlin firmware.
I've installed and used Octoprint on a Raspberry Pi 3 and it all runs properly.
I got as a gift a clone BL-Touch (I believe it's the 3D-Touch) from Amazon.
I was able to install it and it does what it is supposed to...but I didn't get any consistent probes results under Marlin.  
I was about to try playing with the move speed while probing and other various tweaking when I read about Klipper.  
This seems so promissing that I decided to give a try.
I copied over the printer-creality-cr10s-2017.cfg file.  
I also checked the example-extras.cfg file but I'm a bit lost with all these configurations.
I was wondering if someone had a CR-10s configuration file with BL-Touch to share - it would be really appreciated!

FYI, I'm at the base configuration - about to go through the config check document for the base setup.
I haven't even tried to print anything yet since I started down the Klipper path.

Your help is appreciated!
Olivier

RedlineGT, 2019-01-17:
check the klipper Facebook page and search for cr10s and bltouch word combination.

Ozkovacs, 2019-01-17:
Thanks!  I've requested to join the group - waiting for the approval.

Best regards!

Ozkovacs, 2019-01-17:
Closing this as it's not really an issue.  thanks again FeadlineGT for the suggestion!

closed 2019-01-17 by Ozkovacs