Klipper3d/klipper #1125: Correct thermistor type for Creality CR-10 bed and E3D v6 hotend

dovcharov, 2019-01-17:
Good day support.

I see in Marlin that Creality CR-10 (as well as CR-10mini, Ender3 and other printer configs) have EPOCS 100K sensor selected as heated_bed sensor. But in Marlin firmware designed for Creality CR-10 printers, ATC Semitec 104GT-2 is listed as heated bed sensor (option 5). 
What is the correct heated bed thermistor value for Creality CR-10 printers?

Also, can you please confirm, that I should select ATC Semitec 104GT-2 as thermistor for E3D V6 hotend (original)?

lmcbmai, 2019-01-18:
Actually I was going to ask similar questions. I also have original E3D V6 hotend and MK3 heatbead. Both came with sensors that measure 100 kOhm  resistance. I set sensor_type: EPCOS 100K B57560G104F for both then run PID_CALIBRATE and updated pid_Kp, pid_Ki and pid_Kd values. Print quality is good, but I am still unsure that my setup is correct. Could anyone please confirm? Also, does sensor type setting really matter if I run PID_CALIBRATE and set pid values?

Thanks

jakep82, 2019-01-18:
PID really has nothing to do with the type of thermistor.  If you use the wrong thermistor in your config, you will get inaccurate temperature readings.  Different thermistors will have a different resistance at a particular temperature.  For a genuine E3D V6 you should use "ATC Semitec 104GT-2".  Most of the clones I've seen use the "NTC 100K beta 3950" type thermistor.  All available options are listed in [example.cfg](https://github.com/KevinOConnor/klipper/blob/master/config/example.cfg#L151)

dovcharov, 2019-01-18:
I have genuine E3D V6, and manufacturer's manual also states that I need to select ATC Semitec 104GT-2 thermistor. In Marlin firmware I have to set option 5 as hotend thermistor type.
What made me wonder, is default config for Creality CR-10 printer. It's heated bed also comes with ATC Semitec 104GT-2 thermistor, but everywhere, in every config of Klipper you have EPOCS thermistor as default. 
This is only for heated bed. The stock hotend of CR-10 comes with EPOCS thermistor indeed.

cassianoleal, 2019-01-21:
I contributed the CR-10 Mini config but it was a copy-paste of the CR-10 one with changed bed size.

All I can say is that I never had an issue with it, bed heat works as expected but don't really know if the thermistor type is the right one.

TronskiFPV, 2019-01-23:
I'm the latest edits to the ender 3. I have not confirmed temps, but I'm getting the same results as Marlin on the bed and stock nozzle. It has to be within 5 deg on either or things would be wonkey with my ABS. I don't do much PLA but that's all worked as expected too. 
If they aren't correct, they are close enough to have the same settings so far.

KevinOConnor, 2019-02-08:
Okay, I've updated the CR10, CR10S, and CR10mini to use the "ATC Semitec 104GT-2" thermistor, as that is what Marlin uses for these printers.  Commit 161cde4c.

-Kevin

closed 2019-02-18 by KevinOConnor

Deneteus, 2019-08-22:
The thermistor for the heat bed is listed as 100K ohm NTC 3950FB. In the original CR-10 source it uses 1 for both the bed and the hot end.

[image]