Klipper3d/klipper #1127: BlTouch Unknown command:"G29"

Sapherys, 2019-01-18:
Hello everybody,

I'm confused but every time I want to use the G29 command to start the BL Touch Probe, the terminal sais:

> Send: G29Recv: // Unknown command:"G29"

I'm on a Ender 3 with a MKS Gen L and an official Bltouch.

G28 looks to work fine, the pin goes down for a twice probe.

Thank you very much :)

[klippy.log](https://github.com/KevinOConnor/klipper/files/2772961/klippy.log)

Lyr3x, 2019-01-18:
Klipper does not have a G29 G-Code. Check https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md for supported G-Codes.
However what you want to do is a `BED_MESH_CALIBRATE` and then you can save the generated MESH via `BED_MESH_PROFILE SAVE=<name>`
To make it marlin conform (because i am using another printer with marlin and the same slicer settings, i made a macro:

```
[gcode_macro G29]
gcode:
 G28
 BED_MESH_CALIBRATE
 BED_MESH_PROFILE SAVE=p1
 G1 X0 Y0 Z5 F4000
```

Sapherys, 2019-01-18:
You are awson, thank you very much for your fast and complete answer :) it works fine!

Now i could handle the same files with both printers (Marlin/ Klipper) :)

So, if I understand right, the calibration would be saved unter "p1" in your example and then klipper will take these informations automatically while printing?

Lyr3x, 2019-01-18:
If you run a `BED_MESH_CALIBRATE` you need to perform a `SAVE_CONFIG` to write the changes persistent into your configuration. 

Your first mesh is named [default] as you can see in the Terminal. If you run just a SAVE_CONFIG this mesh will always load per default. If you execute a `BED_MESH_PROFILE SAVE=<name>` you add a second mesh which needs to be loaded manual.

jakep82, 2019-01-18:
I'm just clarifying the definition of persistent.  You don't need to save the mesh to make it persistent during a print session.  It only needs to be saved if you want to make the mesh persistent through a restart, and in that case the mesh still needs to be loaded with 

`BED_MESH_PROFILE` LOAD=<name>

Lyr3x, 2019-01-18:
>  You don't need to save the mesh to make it persistent during a print session.
That is not persistency

jakep82, 2019-01-18:
I'm trying to make clear to the OP that saving isn't necessary to use a mesh for printing, and saving the mesh to a config file doesn't mean anything if you don't subsequently load the mesh after restarting the printer.

theopensourcerer, 2019-01-19:
@jakep82 I always assumed if you had a default bed_mesh saved to the config file using the SAVE_CONFIG command it would automatically be loaded when Klipper restarts? This area also contains PID tuning parameters so it would make sense for it be loaded automatically surely? Otherwise I'd better include a LOAD command in my startup g-code somewhere ;-)

Lyr3x, 2019-01-19:
@theopensourcerer Yep that is how it's working. The [default] mesh is loaded every time 👍

NedDarb, 2019-01-21:
Does the default mesh require a call out prior to printing? Or does it automatically load? 

I still run a fresh calibrate before every print, largely because I'm used to requiring it with Marlin. Hesitant to stop as heat effects calibration with inductive sensors and I frequently switch materials.

mocksoul, 2019-01-21:
Marlin didnt required it, just M402 S1 (turn on mesh bed leveling).
In klipper default mesh loaded automatically, you always can check it by BED_MESH_OUTPUT command. It will print if no bed leveling data is loaded and values of bed mesh if they are.

KevinOConnor, 2019-02-03:
I'm going to close this issue as I think the original question was answered.

-Kevin

closed 2019-02-03 by KevinOConnor