Klipper3d/klipper #1128: Python version

BearInThere, 2019-01-18:
As a newbie to python I spent hours yesterday trying to set up an IDE so I could step through the code and understand it.
Having finally realised my problem was that the IDE's I was trying targeted Python3 and that Klipper was written using Python2 things became a lot smoother, and once I realised this I found no shortage of reference here.

If you could add that small bit of info to your overview docs it may save someone else the pain.

Dan B

trevjonez, 2019-01-18:
@BearInThere what did you end up with for your IDE and setup? 

I have gotten pycharm professional working very nicely and the code is all on a remote machine, run and debug included. I was contemplating doing a small tutorial journal entry to help anyone else with the setup. This is my first attempt at python anything so it was a hard earned lesson, but the outcome has been great for productivity.

BearInThere, 2019-01-18:
I’m with you brother
I am keen to share my ignorance.

Kev, your project has obviously evolved to the point of existence.
Should it not be to the point where the curious can ask questions without logging an Issue.

jakep82, 2019-01-18:
You mean through IRC or the mailing list?

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#mailing-list

BearInThere, 2019-01-18:
Trev  I’m hearing you.
I started with VS2017 (where I feel at home). But it was my ignorance of Python as a language that got me into trouble.
Pycharm didn’t fix it.  But it did help me to understand the limitations of my knowledge.
I’ve been paying 15$ a month to ReSharper for the last couple of years without complaint and the fact that Pycharm comes from the same source gives me confidence.

But hey that’s the tool we choose.

BearInThere, 2019-01-18:
Jakep82 
Yeah I am aware of the mailing list 

I was  thinking as the project grows more of a wiki.

Keven has create a brilliant system here and as more people become interested more questions will arise.
They should not be Issues, and the average fool should have a way to ask advice without signing upto a  mailing list.

Hey that's my POV

jakep82, 2019-01-18:
https://github.com/KevinOConnor/klipper/issues/1060#issuecomment-451052072

BearInThere, 2019-01-18:
Cheers.

#1060 (comment)<https://github.com/KevinOConnor/klipper/issues/1060#issuecomment-451052072>

BearInThere, 2019-01-18:
@trevjonez 
Cheers mate but right now I am 100% drunk.
Happy to chat later.

BearInThere, 2019-01-18:
Back to my point
Had you mentioned Python2 in your documentation
I would no be playin 3.6

KevinOConnor, 2019-02-03:
Thanks for the feedback.  I've noted it.  If I get additional queries like this I will add it to the FAQ.

-Kevin

closed 2019-02-03 by KevinOConnor