Klipper3d/klipper #1129: Print is lower than it should, where can I adjust this?

gesmit, 2019-01-18:
I printed the 25mm cube. X and Y were almost perfect, Z was only 24.5mm. How can I fix this?

Lyr3x, 2019-01-18:
If your steps/mm for the z axis is correct (calculate if leadscrew driven) it is mostly due to flow, extruder steps/mm, bed leveling.
I would suggest you start with extruder calibration and then try to get your bed leveling on point. If your cube is about 24.9 - 25.1 you are good.

gesmit, 2019-01-21:
How do I know if the steps/mm for the z axis is correct on my Ender 3?
[stepper_z]
step_pin: PB3
dir_pin: PB2
enable_pin: !PA5
step_distance: .0025
position_max: 240
homing_speed: 5
endstop_pin: probe:z_virtual_endstop
position_min: -5

[extruder]
max_extrude_only_distance: 100
step_pin: PB1
dir_pin: !PB0
enable_pin: !PD6
# This is done in mm/step, not step/mm like other firmwares.
# PS C:> [Math]::Round( (cur_e_step * ((120 - mm_left) / 100)), 6 )
step_distance: 0.009379
#10% increase
# step_distance: 0.008526
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PD5
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA7
control: pid
# tuned for stock hardware with 200 degree Celsius target
pid_Kp: 21.527
pid_Ki: 1.063
pid_Kd: 108.982
min_temp: 0
max_temp: 250
# pressure advance settings
pressure_advance: 1.05
pressure_advance_lookahead_time: 0.015

Paulbsh, 2019-01-22:
Calculate the steps needed for your leadscrews.
https://www.prusaprinters.org/calculator/
But 0.0025 seems OK to me as this would be 400 steps (TR8 leadscrews).

Please dont use the "calibration" cube for checking the dimendional accuracy of your printer. Rather measure if you Z axis moves in fact 25mm when you ask for 25mm

KevinOConnor, 2019-02-03:
I'm going to close this as I think the original question was answered. 

-Kevin

closed 2019-02-03 by KevinOConnor

Nitrosito, 2019-02-06:
Something similar happens to me @KevinOConnor :

Cube withCONFIG basic (delta_calibration)
X 20.66
And 20.31
Z 20.47

cube with extended calibration (print STL, set size .. etc)
X 20.51
And 20.57
Z 20.52

the steps are fine (1/160, pulley 20T in DRV8825) ...
I'll try the Flow to see what happens ¿not possible modify "size" by the other way?

I have an MKS SBASE
a greeting

Nitrosito, 2019-02-06:
Something similar happens to me ...
With the steps OK, extruder OK, flow OK ...

All sides measure practically the same (20.40) should be 20
i recheck all mechanic part tomorrow