Klipper3d/klipper #1130: Cura's Jerk and Acceleration settings

peterlpop, 2019-01-18:
first of all - my prints are coming out great. And fast - Klipper rocks! (set speeds in Cura at 100mm/s on any print and move and I can increase the speed from the printer still). - using Marlin settings (though now I read Repetier might work as well on my Ender 3). 

I am trying to understand how this works and reading the kinematics doc doesn't make it clear to me ... 

**Do the Acceleration and Jerk control settings affect the print or is Klipper using its own settings?**
**Also, the firmware retraction settings in Cura - do I use them or do I tell it to use "firmware retraction"** and use the Klipper config file to set?

mocksoul, 2019-01-18:
Acceleration - yes, Cura makes simple M204 S<ACCEL> gcode which is accepted by Klipper and tuned acceleration of everything (including extruder, so retraction will be changed). 

Jerk - no, klipper has no jerk controls. If you turn on junction deviation in marlin it will not accept jerk tunes anymore as well. Just forget about jerk and dont allow cura to attempt to tune it

FW retraction is not supported by Klipper from the box. But can be easily implement G10/G11 fw retaction commands in printer.cfg. They will differ from marlin implementation, though. Because G10 G10 G10 G11 in marlin will make just 1 retaction and 1 prime, but macro will make 3 retractions and 1 prime. But I guess cura is smart enough to not do a bunch of consecutive retracts without recovering :)

The main reason G10/G11 should be made is to able to tune retraction settings on the fly while printing. GCODE size reduction is marginal here.

peterlpop, 2019-01-18:
> Acceleration - yes, Cura makes simple M204 S gcode which is accepted by

This completely answers my questions about the first two settings! Thanks.

That is exactly the reason I would like to do G10/G11 - I use multiple
computers and some settings in the firmware config file would be great.

Can you give me a sample of the G10/G11  as a macro ... I am sure a sample
would help me figure out how to put it in the config file.

And how would I tune it on the fly - saving the config file resets the
host/firmware?

mocksoul, 2019-01-18:
G10/G11 -- as I said you are unable to tune them in the fly during print in klipper, coz it requires to change printer.cfg as restarting klippy as of now.

I didnt tried to set them by myself, but it should be simple. Take a look here for examples #824 =)

peterlpop, 2019-01-18:
Got it, Though I will have to confirm before using this code that Cura is using G90 absolute Extrusions during normal operations. Just to confirm, variables (that can be changed in Terminal, don't work as of now in Macro code (and I am definitely not going to attempt Python functions for now)?

[gcode_macro G10]
gcode:
    G91
    G1 E-2.5 F4000
    G90

[gcode_macro G11]
gcode:
    G91
    G1 E2.5 F500
    G90

mocksoul, 2019-01-19:
> Cura is using G90 absolute Extrusions 
That is tunable in Cura settings, "Relative extrusion", at least in 4.0-beta. By default, yes, cura use absolute. Using relative extrusion allows ease preprocess of gcode file and replacing cura's retract with custom one :)

KevinOConnor, 2019-02-03:
I'm going to close this as I think the original question was answered.

-Kevin

closed 2019-02-03 by KevinOConnor