Klipper3d/klipper #1131: Feature Requests - Dual X axis

Neukam, 2019-01-19:
I am interested in trying Klipper on a dual x carriage printer and there are a few features that I would like to see implemented, if possible.  I don't have any programming skills, but I would be happy to help in testing.

1.  duplication mode - T1 moves to T0 position plus half of the usable bed size and moves in sync with T0,  Extruder 1 moves in sync with extruder 0

2.  Mirror duplication mode - T1 moves to bed size minus T0 position, then moves in the opposite direction, in sync with T0, but limited by a minimum separation distance between T0 and T1  Extruder 1 moves in sync with extruder 0

3.  Fan speed follows active extruder - Each extruder should have its own fan output and during tool change, the fan on the newly selected extruder should be turned on at the previously set speed and the fan on the parked extruder should be turned off.  If duplication mode is active, both fans should operate together.  This feature could be useful on a dual nozzle printer with a single carriage if someone wanted a dedicated fan for each nozzle, but would be especially useful on dual x carriage printers

KevinOConnor, 2019-01-21:
Interesting.  The above would require a developer to implement, however.  It is not something I personally will have time to look at in the short term.

-Kevin

bruce356, 2019-01-31:
I agree that option 1.  would be very handy and a nice addition.
Option 2.  a model can so easily be mirrored in cad or slicer not sure if its worth the programming effort in firmware.
Option 3. I feel this really is not necessary, a fan draws virtually nothing and the second fan can be used to help cool a model from both sides if the carriage is strategically placed.
Just my thoughts Regards - bruce

Neukam, 2019-01-31:
The purpose of mirror mode would be to print both the part that was modeled and sliced, and it's mirrored equivalent and the same, using both carriages. 

Fan control is needed because multiple fans are not well supported in slicer software.  Having a fan running on a parked extruder can cause uneven heating if the bed and uneven cooling off the parts you are printing.

bruce356, 2019-01-31:
@Neukam  sorry made a mistake in my comments regarding mirror option (late night).
As for fans, I use blowers with directed air flows so do not suffer from air directed onto the bed (as may be the case with axial fans), on my IDEX printer when the carriage that is not printing is parked, its located 40mm off the side of the bed. I have never had a problem with uneven cooling due to parked fan remaining on. Your situation may be different to mine.
regards - bruce

KevinOConnor, 2019-02-18:
Thanks all.  I'm going to mark this as inactive as it doesn't look like this item is being worked on right now.

-Kevin

closed 2019-02-18 by KevinOConnor

nailgilaziev, 2020-05-19:
Did I understand correctly that IDEX on Klipper currently not available?

scheerjacob, 2020-08-05:
If anyone is using a Tenlog or Hictop Dual Extruder, can you please share with me your config files? I am trying to get Klipper to work with Tenlog D3 Pro and it would be very helpful to look at someone's config file for that. 
@Neukam @bruce356 @nailgilaziev
@KevinOConnor

boubouexpress, 2021-04-20:
Bonjour
Klipper est vraiment super
Dommage qu'il manque les deux fonctionnalité primordial pour une idex
 une imprimante qui va devenir une référence ,on le voit bien sûr youtube ça fait un carton .
Si klipper ne fait pas ces options, il se faire des portes.