Klipper3d/klipper #1133: Possible? Adjust dual z-axis motors by two top end stops?

HWiese1980, 2019-01-19:
Hi everyone,

my printer uses two motors to drive the z-axis by means of two lead screws. The motors aren't connected mechanically other than the beam where the extruder is mounted on. Because of that it's possible that the motors get "out of sync" so to speak. Not during a print but while they are disabled. 

I would like to be able to re-adjust them by means of two end stops at the top of the printer. An adjustment would run by moving the z axis up until one end stop triggers and then continuing movement of the remaining motor on the untriggered side until the second end stop triggers as well. Maybe followed by another, slower reference run. 

Is that already possible with Klipper maybe by using some of the custom pin settings, gcode macros and stuff? Or would it require some additional coding? 

What would be necessary?

jakep82, 2019-01-19:
This is already supported.  Simply configure stepper_z and stepper_z1, and then assign the appropriate endstop pins to each stepper.

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L339

HWiese1980, 2019-01-20:
I forgot to mention that I have a probe for ABL. That's why I asked to have them at the top of the printer. Sorry...

Is it also supported to have both?

KevinOConnor, 2019-01-25:
Yes.  See the description of the [probe] config section in config/example-extras.cfg - it is separate from the stepper_z configuration and its use is not exclusive.

-Kevin

KevinOConnor, 2019-02-03:
I'm going to close this as I think the original question was answered.

-Kevin

closed 2019-02-03 by KevinOConnor

HWiese1980, 2019-02-03:
Well, yes, the question **itself** is answered: it can be done. 

The obvious follow-up question though is not - at least not to my liking: **How?** 

Because of the probe I have an `endstop_pin: probe:z_virtual_endstop` entry in my `stepper_z` section. How can I combine that with two additional *real* endstops switches that are each associated to a single stepper of the z-axis?

Sorry, I only have a very unreliable internet connection where I currently am and a bunch of stuff on my todo list (we have moved recently) - because of that I almost forgot about this thread. So, thanks for reminding me.

HWiese1980, 2019-02-03:
The thing is, it would require at least one z_stepper to have multiple endstops, probably both: one `z_virtual_endstop` and one physical endstop pin. It furthermore would have to have a homing command that homes to z-max, not z-min as with the probe. 

I'm not sure if that's really already supported. If it is, I'd really like to know how to configure it. I can't really see it in the example config files.

HWiese1980, 2019-02-03:
Okay, yes, that homing procedure is clear and makes sense. Is it still possible to use ABL with the probe in that configuration? Because currently essentially it's supposed to be three endstops for the z-axis. Two at z-max (one for each stepper) in order to level the z-beam, and one endstop for z-min (the probe) to level the print bed. I don't have any other, physical endstop(-switch) at z-min because I want the printer to be able to do ABL.

schnello, 2019-09-16:
Hello @HWiese1980 

Yes this is possible but from my view only a work around.
I have the same config. (i3 mega)
1 z probe for ABL and two stepper motors with two mechanical endstops.

So my config is:
[stepper_z]
step_pin: ar46
dir_pin: ar48
enable_pin: !ar62
step_distance: 0.002519031575
endstop_pin: probe:z_virtual_endstop
endstop_pin: ^!ar18
position_endstop: 0
position_min: -1
position_max: 205
homing_retract_dist: 2.0
homing_speed: 6.0
second_homing_speed: 1

[stepper_z1]
step_pin: ar36
dir_pin: ar34
enable_pin: !ar30
step_distance: 0.00247085
endstop_pin: ^!ar43

With this config a can:
Home the z axis .... the two mechanical endstop are used and for ABL i use the probe.
But... dont use this config to home with the probe. I performed many tests and i came to the to the conclusion: With this config the zprobe offset is ignored

grykom, 2019-09-24:
> Okay, yes, that homing procedure is clear and makes sense. Is it still possible to use ABL with the probe in that configuration? Because currently essentially it's supposed to be three endstops for [...]

hi xxx, did you deal with this problem? I have exactly the same idea to do it just like you and the only thing that comes to my mind is to give physical endstops lower than BLtouch and turn off z_virtual_endstop for the moment of calibration.

schnello, 2019-09-24:
@grykom 
Please see https://github.com/KevinOConnor/klipper/issues/1376