Klipper3d/klipper #1135: generic-printrbot config burned out my stepper drivers

drphil3d, 2019-01-20:
I used the config generic-printrboard on my RevF and it burned out two of my stepper drivers so far. 
I don’t know what scale 2.327 refers to I had decreased the number down to 1 but the motors where very hot so I just put it back to the default value in the config, then halfway into a print my drivers failed.

Can anyone help enlighten me on this ? 
Thank you

[mcp4728 stepper_current_dac]
i2c_mcu: eux
scale: 2.327
channel_a: 1.3 # extruder_4
channel_b: 1.1 # extruder_3
channel_c: 1.1 # extruder_2
channel_d: 1.1 # extruder_1

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

KevinOConnor, 2019-01-20:
The mcp4728 parameters are documented in the config/example-extras.cfg file.  One would not typically change the scale value (unless the hardware was altered, or it was felt the existing calculation was incorrect).  Instead, one would normally modify the channel values to set the desired amps for each stepper.

-Kevin

drphil3d, 2019-01-20:
Oooh now that you mention that I see what your talking about, I feel lightly stupid.
This should set the drivers to 500ma 
channel_a: .5 # extruder_4
channel_b: .5 # extruder_3
channel_c: .5 # extruder_2
channel_d: .5 # extruder_1