Klipper3d/klipper #1136: MKS sbase tmc2208

lightface79, 2019-01-20:
Hi.
New to Klipper and so far i love the system and the ease of config. 
Big THUMBS up to Kevin & everyone else.

Now the problem i have!!
Running a MKS sbase 1.3 card lpc1768
On this i run 4 TMC2208 drivers.

now i run just 1 tmc2208 for error and debugging
What i cant wrap my head around is witch pins im going to use.
I had a good connection to 1 driver.
Where i used Aux-1 P0.2 & P0.3
Using 2 cabel rx/tx Tx 1kohm 

Tried a bunch of different pins in Exp1 & 2
im going totaly bunkers soon =(

then i saw in the doc for tmc2208 here on kevins github
that u can run only 1 cabel  to the uart from the mcu of the lpc1768
dont u need to have 1 cable from the driver that splits in to two rx/tx  &tx 1kohm?

Please help.
Best regards Janne /Sweden

KevinOConnor, 2019-01-21:
Klipper supports using just one pin for the tmc2208.  (Klipper is able to switch the pin between input and output at runtime - other boards/firmware may not be capable of doing that so Trinamic documents a two pin setup.)  As for which pins to use - I don't have much advice there.  Any general purpose pin (that isn't in use for some other purpose) should work fine.

-Kevin

lightface79, 2019-01-21:
ok... so 1 cable to eany interuptebal pin on the lpc176x 
do i need a 1k resistor on that cable or just plain straight wire to the 2208

KevinOConnor, 2019-01-21:
No resistor is needed.  Any gpio pin works.

-Kevin

lightface79, 2019-01-21:
Think i got it working...   it dossent crash and disconect when i run status.  
and im getting the hed temp as well 

but hoe do i use DUMP_TMC  ?  
or do i need to add somthing in the config file for that to work to read the tmc2208

lightface79, 2019-01-21:
No worrys...   i found out what to do =)  

DUMP_TMC STEPPER=stepper_a

Send: DUMP_TMC STEPPER=stepper_a
Recv: // GCONF:          000001c4
Recv: // GSTAT:          00000001
Recv: // IFCNT:          00000012
Recv: // OTP_READ:       0000000d
Recv: // IOIN:           20000341
Recv: // FACTORY_CONF:   0000000d
Recv: // TSTEP:          000fffff
Recv: // MSCNT:          000002e8
Recv: // MSCURACT:       01dc010b
Recv: // CHOPCONF:       14030053
Recv: // DRV_STATUS:     80060000
Recv: // PWMCONF:        c80d0e24
Recv: // PWM_SCALE:      00000007
Recv: // PWM_AUTO:       000e0024

is the anwsers always in HEX i belive?

mocksoul, 2019-01-21:
> is the anwsers always in HEX i belive?
> is there a bulit in translator for this?

Take a look in this comment: https://github.com/KevinOConnor/klipper/issues/196#issuecomment-455785459

closed 2019-01-30 by lightface79