Klipper3d/klipper #1137: Hot End staying above set temp

chronitale, 2019-01-21:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2777437/klippy.zip)
[image]

My hot end temperature heats up fine but stays 4-10°C above set temp. I have ran PID tune a few times and no change. I'm using a PT1000 thermistor from TriangleLabs and have set adc_temperature that I found from another post.

KevinOConnor, 2019-01-21:
I, personally, do not have familiarity with the PT10000.  If you have a link to a data sheet and a link to the hardware schematic that you are using to wire the PT10000, then I can try to take a quick look at it.

-Kevin

chronitale, 2019-01-21:
Pulling this info from their sale page: [Here](https://www.aliexpress.com/item/Trianglelab-PT1000-Thermistor-Cartridge-for-3D-printer-E3D-Volcano-v6-heater-block-UPTO-450C-PEEK-PEI/32859917151.html?spm=a2g0s.9042311.0.0.22b24c4d1Go8F0)
Here is the resistance chart http://www.hayashidenko.co.jp/en/info14.html
Here is the settings for Marlin:
Marlin:
Default 4.7Kpullup resistor please choose sensor '1047'
If you have replaced pullup with 1K resistor please choose sensor '1010'
Configuration.h:
*1047 : Pt1000 with 4k7 pullup
*1010 : Pt1000 with 1k pullup(non standard)
#define TEMP_SENSOR_0 1010
#define TEMP_SENSOR_1 0
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_4 0
#define TEMP_SENSOR_BED 1
thermistortable_1010.h:
// Pt1000 with 1k0 pullup
const short temptable_1010[][2] PROGMEM = {
  PtLine(    0, 1000, 1000)
  PtLine(  25, 1000, 1000)
  PtLine(  50, 1000, 1000)
  PtLine(  75, 1000, 1000)
  PtLine(100, 1000, 1000)
  PtLine(125, 1000, 1000)
  PtLine(150, 1000, 1000)
  PtLine(175, 1000, 1000)
  PtLine(200, 1000, 1000)
  PtLine(225, 1000, 1000)
  PtLine(250, 1000, 1000)
  PtLine(275, 1000, 1000)
  PtLine(300, 1000, 1000)
  PtLine(325, 1000, 1000)
  PtLine(350, 1000, 1000)
  PtLine(375, 1000, 1000)
  PtLine(400, 1000, 1000)
  PtLine(425, 1000, 1000)
  PtLine(450, 1000, 1000)
};
thermistortable_1047.h:
// Pt1000 with 4k7 pullup
const short temptable_1047[][2] PROGMEM = {
  // only a few values are needed as the curve is very flat
  PtLine(    0, 1000, 4700)
  PtLine(  50, 1000, 4700)
  PtLine(100, 1000, 4700)
  PtLine(150, 1000, 4700)
  PtLine(200, 1000, 4700)
  PtLine(250, 1000, 4700)
  PtLine(300, 1000, 4700)
  PtLine(350, 1000, 4700)
  PtLine(400, 1000, 4700)
  PtLine(450, 1000, 4700)          
};

I tried the following but the printer would not connect.
[thermistor PT1000]
temperature1: 0
resistance1: 100
temperature2: 100
resistance2: 138.50
temperature3: 200
resistance3: 175.84

chronitale, 2019-01-21:
Oh fudge I did put the decimal in wrong spot. I put in the following and get this error after restart.

[thermistor PT1000]
temperature1: 0
resistance1: 1000
temperature2: 100
resistance2: 1385.10
temperature3: 200

Recv: // Internal error during connect.
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Printer is halted
Recv: // Klipper state: Not ready
Recv: !! Internal error during connect.

KevinOConnor, 2019-01-22:
I added support for using temperature/resistance measurements to adc_temperature (commit 7327394c).  You'll need to pull the latest version of the code to use this feature (see the FAQ for details on upgrading).  After upgrading, see the updated config/example-extras.cfg file for details on setting temperature and resistance measurements in adc_temperature.

-Kevin

chronitale, 2019-01-22:
[image]
[klippy.zip](https://github.com/KevinOConnor/klipper/files/2782461/klippy.zip)

I'm still running into the same issue and ran PID twice. I'm using a 50watt heater in hotend, could that be a problem? I wouldn't think so because the PID should be taking care of that.

KevinOConnor, 2019-01-23:
That's odd - I've not seen a heater actually heat that fast.  Could the sensor be too close to the heater?  As it stands now, the temperatures are too volatile to get a good sync.

You could try setting the extruder max_power=0.2 to see if that helps.  You may also want to use an external temperature probe to verify that the reported temperatures are accurate.

-Kevin

chronitale, 2019-01-23:
[image]
Much better! I have a copper Volcano hotend and 50w heater so I could try printing with 1mm nozzle at higher speeds. I am assuming that the heat transfer is too great for the PID to maintain steady temps.

Thank you for your quick help with this issue.

closed 2019-01-24 by chronitale

elitecapo, 2021-04-18:
Hey I am literally having the same issue, what was the PT1000 senor code you used in the end?

chronitale, 2021-04-19:
I had to reduce the power to the heater as well. Here is the code I'm using.
[extruder]
heater_pin: ar10
max_power: 0.18
sensor_type: PT1000

[adc_temperature PT1000]
temperature1: 0
resistance1: 1000
temperature2: 100
resistance2: 1385.10
temperature3: 200
resistance3: 1758.60