Klipper3d/klipper #1140: wiring guide to convert arduino mega with ramps to use klipper with raspberry

rizpardazan, 2019-01-22:
hello all
is there any wiring guide to convert an arduino mega with ramps based printer to use klipper with raspberry?
i mean wiring guide from raspberry gpio to ramps board.

and another question:
is it possible to use klipper with ramps without using arduino ?

theopensourcerer, 2019-01-22:
Not sure I follow. To use Klipper with a standard RAMPS setup you connect the USB from the Arduino to the RPi and then follow the Installation instructions [1] to compile and replace the firmware.

[1] https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md

I think the answer to your second question is no.

theopensourcerer, 2019-01-22:
Please read the comprehensive documentation... It explains this.

https://github.com/KevinOConnor/klipper/blob/master/docs/Features.md

rizpardazan, 2019-01-22:
> > and another question:
> > is it possible to use klipper with ramps without using arduino ?
> 
> Seems like you didnt understand how Klipper works.
> The Ramps is more or less an adapter board that maps the arduino pins to the drivers (and a few other peripherals). The Ramps itself has no logic, nothing to interpret the signals coming from the Pi.

I know that.
But what i confused is that mega 2560 can handle microsteps at about 20kH
Then hiw klipper can speed up it ?
I think that if we can use klipper in raspberry without an arduino and only map gpio to ramps boarf pins we can get more speed because of 64bit procesor in raspberry

theopensourcerer, 2019-01-22:
@rizpardazan 

Please read the comprehensive documentation...

jakep82, 2019-01-22:
This has been discussed several times here already.  You would need to make significant changes to Klipper to make this work, and it wouldn't work very well.  Do some searching on the internet and you'll see that a lot of people have had this same idea, and overcoming the technical limitations (jitter is a major problem) isn't worth the effort when an inexpensive 32 bit MCU is capable of generating half a million precisely timed steps per second with already available software.

https://github.com/KevinOConnor/klipper/issues/925#issuecomment-445556483
https://github.com/KevinOConnor/klipper/issues/831#issuecomment-434095394

closed 2019-01-22 by rizpardazan