Klipper3d/klipper #1145: Adding support for importing .cfg files into 'printer.cfg' ?

JohnEdwa, 2019-01-23:
Examples speak louder than words, so lets assume you could include files and Klipper allowed you to redefine variables. A full printer.cfg could then look like this:

```
[include] 
 klipper/config/printer-creality-ender3-2018
 klipper/scripts/gcode_m300.cfg 
 custom_lcd_menu.cfg # complete rewrite of the menu config ~1200 lines long.

# reverse extruder and calibrate steps
[extruder]
 dir_pin: PB0
 step_distance: 0.011410184

[mcu]
serial: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A107FTEV-if00-port0
```

But to be honest, I just want the ability to split my gcode scripts and lcd menu modifications to separate files as my printer.cfg is currently ~1200 line long and it's annoying me. Wouldn't need any redefines or anything, just allowing adding files to be appended to printer.cfg.

theopensourcerer, 2019-01-23:
+1
I'd like to see this too.

Personally I'd opt for something like the config.d directory structures of debian et al, where all files in it get included. Perhaps, like udev rules and such like, each file in here could be prefixed with a number so it gets loaded in the right order if that would be necessary.

Lyr3x, 2019-01-24:
That way it'll be possible to get basic board and lcd definitions a user dont need do bother with.

theopensourcerer, 2019-02-03:
I'd be happy to have a go at this. I'm not that experienced with Python but it might be good as a learning exercise for me. 

I guess the first question would be how to approach this? 

My personal preference would be to do something like this (pseudo code):

if(printer.cfg in home dir):
    load that as normal
elseif(printer.cfg.d directory exists):
    load all files in directory, in order

With the config files being like: (names are arbitrary, number-prefix determines order to load and parse)

1-base.conf
10-kinematics.conf
20-extruder.conf
30-bed.conf
90-gcodes.conf

etc. etc.

Anyone got any other suggestions / alternative mechanisms? 
Anyone have any issues with the approach above?

technoplunk, 2019-02-07:
I would like to add that in most .d config schemes I have seen that as long as primary config exists it is loaded. The printer.cfg.d/ files are loaded/appended to it after and can override the printer.cfg. So the logic would be more like:

```
if(printer.cfg in home dir):
    load that as normal
if(printer.cfg.d directory exists):
    load all files in directory, in order appending/overriding printer.cfg.
```

theopensourcerer, 2019-02-08:
@JohnEdwa This is a good point. Perhaps an "include" syntax of some sort would be a more obvious and easily controllable approach. 

The way the debian/apache config works is another interesting way (only files with a .conf extension are "included" and you control the inclusion using a simple command line tool to enable or disable config options, i.e. individual files which creates and breaks symbolic links. I can again see this might again not be a particularly obvious approach for non-linux users but I quite like the concept.

I think coming up with a sensible plan before actually trying to write any code is the right way to do this. Your comment about Octoprint is also valid. In the Octoklipper plugin, there is a feature now to edit the config file (although I think it's still a bit broken due to the SAVE_CONFIG data) so ensuring that whatever approach is selected can be supported by a control/configuration tool in either Octoklipper or any other plugin should also be a highly desirable feature IMHO.

Papsnoek, 2019-02-08:
for Me, I prefer @JohnEdwa Approach. It makes way more sense to use a "Include" kind of commands, as then you can easily switch out which Files/Configs to include into the Main Config. Thereby allowing you to setup different scenarios, and easily switch between configs without having to Copy/Move Files between multiple directories, etc. Also, to the General User, it would be more understandable as to WHAT you are going to include, and what ORDER it will be included.

This approach would also solve Issues #1179  as you can have a Main Config, and just change which file to Import which contains the Actual Config needed.

theopensourcerer, 2019-02-08:
I wonder if there is any mileage in having an [include ...] section type approach? 
1. It would be consistent with the way the rest of the configs work
2. Potentially you could then do something like (just an idea)

[include fast_setup]
active: true
directory: fast_system.d
file: x_and_y_steppers.conf
file: extruder.conf
file: gcodes.conf
...

Then a "extras" module could be written to handle the loading and parsing etc...

I was trying to think about if this could be an easy way to have different include sections

[include quality_setup]
active: false

And have an "active" flag or something which would make it easy to switch configurations around? Not sure if am over-thinking this or not?

KevinOConnor, 2019-03-04:
I'm going to mark this as inactive as it doesn't appear it is currently being worked on.

-Kevin

closed 2019-03-04 by KevinOConnor

KevinOConnor, 2019-05-26:
Config file include support is now in Klipper.

-Kevin

PR #1359 (2019-03-23): configfile: Add "include" support
Allows configuration files to include other configuration files using `[include filename.cfg]` syntax.  Klippy loads include files in the position of the include header; subsequent definitions override included values.

Supports wildcards (e.g. `[include macros/*.cfg]`).  Allows included files to include other files but blocks recursion.

Resolves #1145

Example `printer.cfg`:

```ini
# Load standard simulator config
[include klipper/config/avrsim.cfg]

# Make it faster
[printer]
max_velocity: 750
max_accel: 4500

# Add our macros
[include macros/*.cfg]
```