Klipper3d/klipper #1146: RGB Led Strip control with Raspberry/klipper?

MiTsCh13, 2019-01-23:
Hey!
I want to control an RGB Strip with my raspberry pi (which is running octoprint + klipper). I have connected the strip to the raspberry (physical) pin 13, 15 & 16 (R, G and B) and tried to control it with the LedStripControl plugin, but it does not work.

Is it even possible to get this working? If so, how do I have to configure this to be able to shut the leds down and change the colour..?

Thanks in advance!

jakep82, 2019-01-23:
This would be a question for the developers of that plugin.  Klipper is not able to control the gpio pins of the Pi.

Lyr3x, 2019-01-24:
Did you connected the strip directly to the pi? This will not work. You need MOSFET's for each channel to control them (they are PWM controlled). After that you can setup the pins in the Octoprint Plugin and control each channel with M150 RGB(W)

MiTsCh13, 2019-01-24:
> 
> 
> Did you connected the strip directly to the pi? This will not work. You need MOSFET's for each channel to control them (they are PWM controlled). After that you can setup the pins in the Octoprint Plugin and control each channel with M150 RGB(W)

Hey Lyr3x!
No, I did not connect them directly. I have wired MOSFETs for each channel, but the M150 did not work. (On the website for the LedStripControl it also says that the M150 command is for marlin, not for klipper)

Lyr3x, 2019-01-24:
No there is a notive that it has the same syntax as the marlin M150 but the command intercepts octoprint and control the GPIO instead. Check your wiring. I just set this up and it is working 👍

Lyr3x, 2019-01-25:
I can't see a miswiring... What kind of Mosfet do you use? 
Are u sure you didn't swapped anything?

MiTsCh13, 2019-01-26:
As shown in the picture, I used BuZ71A Mosfets (here is the datasheet: http://pdf1.alldatasheet.com/datasheet-pdf/view/22132/STMICROELECTRONICS/BUZ71A.html)
And yes I am sure that I did not swap anything..

Lyr3x, 2019-01-26:
At least these Mosfets are not optimal to use with the Pi. The gate max threshold is 4V but your pi is 3.3V on High. That way your LEDs won't be very bright. But I think there should any feedback though.

MiTsCh13, 2019-01-26:
Okay so I'll look out for other mosfets.
But isn't it strange that the LED strip lights white when I turn on the 12V PSU even though the output pins are on 0V?
I thought that the strip starts lighting when I turn on the 12V AND turn the R, G and B on high?

Lyr3x, 2019-01-26:
Yes it is because that means that the gate is open what can only be if you wired the 12v to the gate pin.

kingfisher1234, 2019-01-27:
I've just ordered a RGB LED amplifier and was going to follow this link as an easy way of wiring the LEDs to the PI

https://www.instructables.com/id/Easiest-RGB-LED-Strip-With-Raspberry-Pi/

KevinOConnor, 2019-02-08:
Okay, I'm going to close this as it sounds like a solution was found (and it was via OctoPrint and not Klipper).

-Kevin

closed 2019-02-08 by KevinOConnor