Klipper3d/klipper #1150: When hotend is done heating, it stops heating, without giving me an error and slowly cools down.

CharlesStArnault, 2019-01-24:
The title pretty much explains my issue.

[my log.zip](https://github.com/KevinOConnor/klipper/files/2793955/my.log.zip)

jakep82, 2019-01-24:
Your hotend temps are also fluctuating a lot.  This looks like you might have a noise issue with your thermistor wiring.

CharlesStArnault, 2019-01-24:
I've been messing a lot with my pid settings, that might be the cause. So you are telling me that because my temps are fluctuating a lot it's causing the hot end to cool down without any error message in octoprint?

jakep82, 2019-01-24:
I'm not sure why you aren't getting an error in Octoprint, but if you look at your log you will see "Heater extruder not heating at expected rate".  That actually means the error is coming from your extruder, but I think you'll have similar isssues with your bed.  If you were to type status in the terminal after this happens, it should show you this error.

CharlesStArnault, 2019-01-24:
Alright, ill be doing tests and getting back with you. Thanks for your help!

KevinOConnor, 2019-01-25:
The log seems to to indicate that your power supply is not powerful enough to heat both your extruder and bed at the same time.  I recommend:
1. Set a max_power in your [heater_bed] section to reduce the amount of power sent to the bed.  (For example, set max_power=0.5 .)
2. If possible, tune PID for the bed and avoid using the watermark heater.
3. Always heat your bed to the target temp prior to heating your extruder.
4. It may not be viable to heat your bed to a temperature of 85 degrees without hardware upgrades.

-Kevin