Klipper3d/klipper #1153: STM32 memory requirements / specs

jammi, 2019-01-25:
What are the memory  requirements for STM32F3 series MCUs? AFAIK F3 should be basically backward-compatible with F1, but I didn't find any information on what the minimum Flash and RAM requirements of Klipper are. Would it for instance work with something that has just 64k Flash and 16k SRAM for instance?

KevinOConnor, 2019-01-27:
As far as I know, the STM32F1 code is coded for 16K chips.  Klipper itself can run on chips with as little as 1K of ram (though it really wants at least 4K).  The flash size varies a bit depending on architecture, but it is typically around 16K in size.  I don't know what the STM32F3 would require.  I suspect supporting the STM32F3 would require code changes.

-Kevin

jammi, 2019-01-29:
Ok, I had planned to run a RAMPS with one or two of the three ST Nucleo boards I have, once the shipping of one of the supported controllers arrive in order to get started with Klipper.

KevinOConnor, 2019-02-08:
Okay, I'm going to close this as I think the original question was answered.

-Kevin

closed 2019-02-08 by KevinOConnor