Klipper3d/klipper #1154: "BLTouch failed to verify sensor state" on "G28 Z"

stemaDev, 2019-01-25:
"Send: G28
Recv: !! BLTouch failed to verify sensor state" (X and Y homes, goes to center, but gives this error message when it should start going down)
"pin_up" and "pin_down" works ("control_pin" should be fine)
"QUERY_PROBE" shows "opened" and "TRIGGERED" correctly ("sensor_pin" should be fine)
Because of the things above, I would also think that I installed the bltouch right and that it works. It worked before on a tevo tarantula with an earlier klipper, now it's on an ender 3 (stock board) with the latest klipper. It's a clone, called "3d touch".

gerrylenz, 2019-01-26:
My workaround is

[bltouch]
...
pin_up_touch_mode_reports_triggered: False
...

stemaDev, 2019-01-26:
> pin_up_touch_mode_reports_triggered

It starts to go down, but the pin isn't getting deployed that way (and light stays on). Also tried pin_up_reports_not_triggered and combinations, but nothing yet worked. Thanks though (Y)

fredizzimo, 2019-01-26:
Many people have better luck with my version, https://github.com/KevinOConnor/klipper/pull/982. There are instructions near the bottom for how to test it out, so could you please test that instead?

So far I'm not aware of any clone that doesn't work with that, at least if you set `test_sensor_pin` to False. There might be some non-working configuration, but no one has provided proper logs so I can see if if something really is wrong or not. Note that the `test_sensor_pin` setting only tests for very specific cabling errors, and only works one genuine probes, most clones don't support it. Even if you have it disabled, the built in error checks are performed, which can detect a much wider range of errors. 

I also recommend my version if you care about more robust behaviour. These are some tests that I did with a genuine BLTouch Smart https://github.com/KevinOConnor/klipper/pull/1091#issuecomment-453749542, on the version that we merged to master, as you can see there are a lot of very basic issues.

stemaDev, 2019-01-26:
"pin_up_touch_mode_reports_triggered: False" did work, maybe the cables went loose or maybe the "False" is case sensitive, but it works now, thank you!

closed 2019-01-26 by stemaDev