Klipper3d/klipper #1157: delta_calibrate stuck in "Working oncalibration"

cmock, 2019-01-26:
I'm just mechanically rebuilding my Kossel Mini and have been issuing quite a few DELTA_CALIBRATE commands today.

The last one of those, instead of almost immediately outputting the parameters after probing, just said "// Working on calibration..." over and over again. 

Fine, I think to myself, the optimization algorithm is running wild, but no, "top" shows no CPU usage for the klippy process...
[klippy.log](https://github.com/KevinOConnor/klipper/files/2799074/klippy.log)

KevinOConnor, 2019-01-27:
FYI, I saw this recently on my delta printer as well.  My best guess was that it occurred because I ran detla_calibrate multiple times prior to finishing the first attempt.  I'll add it to my list of things to look at.

-Kevin

Nossolov, 2019-02-15:
Same for me.
Once with advance calibrate, and yesterday with normal manuel calibration.
I thing (but not sure) that one calibration point (7 points for me) was not good and far from real (Maybe 1mm, my fault...).
Reboot the raspberry, calibrate again, and everything fine.

pappabj0rn, 2019-08-14:
Ran into this yesterday too.
Was running a manual calibration and accidentally hit accept (set up as macro button) just after moving to a new point (so 5 or so mm up of the bed). It didn't like that and I had to abort and redo it, which den got stuck in the working on calibration loop.

KevinOConnor, 2019-08-16:
@pappabj0rn - are you on a recent version of the code?  (I was thinking this might have been fixed with some reactor changes a few weeks ago, but I guess not.)

-Kevin

pappabj0rn, 2019-08-16:
Yeah, just installed it this tuesday, I think.

KevinOConnor, 2019-12-13:
FYI, I did find and fix an error in this area of the code (commit 139274f0).  If anyone sees this error again, please report it (and provide the klipper log file).

-Kevin

Nossolov, 2019-12-13:
Ok, thank you. 
Everything works so great that i'm afraid by doing an update now ! 
May be for the next printer to come ? 
Really a great job. 

----- Mail original -----

KevinOConnor, 2019-12-15:
Alas, it's definitely still broken.  I added a delta_calibrate test case, and now the automatic build will randomly fail sometimes..

-Kevin

Nossolov, 2019-12-15:
Of course it's sad, but everything else works fine and manual calibration is easy and good. 

No chance i change Klipper for another one ! 

----- Mail original -----

Alkarasu, 2019-12-16:
> FYI, I did find and fix an error in this area of the code (commit [139274f](https://github.com/KevinOConnor/klipper/commit/139274f0bbb0cd36e4bf1b858ea14da3b7859c2a)). If anyone sees this error again, please report it (and provide the klipper log file).
> 
> -Kevin

Got hit by it today. Here's the log.
[klippy.log](https://github.com/KevinOConnor/klipper/files/3969668/klippy.log)

KevinOConnor, 2019-12-17:
I found and fixed an error that could cause these issues (commit abf49349).  Hopefully it is now fixed.  If anyone runs into this with the latest code, please report and attach the log.

-Kevin

KevinOConnor, 2020-02-14:
I think this has been fixed.  The DELTA_CALIBRATE command has been in the automatic build tests and there hasn't been a build failure since abf4934.

-Kevin

closed 2020-02-14 by KevinOConnor