Klipper3d/klipper #1158: axis stop moving after change step_distance

Astrias57, 2019-01-26:
Hello,

I calibrated my printer with correct step distance but had an issue after that.

Initial step distances were 0.0125 for X and Y and 0.0025 for Z and I did not have any issues. 
After changing these values to 0.01246, 0.01249 and 0.002495 respectively, Y axis and then X axis stop moving correctly after 1h of printing and i obtain the results shown in atteched picture.

Could you help me to resolve this issue please ?

[log_and_photo.zip](https://github.com/KevinOConnor/klipper/files/2799171/log_and_photo.zip)

KevinOConnor, 2019-01-27:
Are you using tmc2208 drivers?  If so, be sure to see issue #196.

-Kevin

Astrias57, 2019-01-27:
Thank you for the fast answer. Yes i'm using TMC2208 both for X and Y. I will follow Issue #196 .

It's strange that a small change in step distance cause this issue. Without changing this I can print during 6h without stepper shutdown. I never try to print for a longer duration so perhaps the moter would stop after a longer printing time.

KevinOConnor, 2019-01-27:
This in line with our understanding of the problem.  By changing to an "unusual" step distance it made it much more likely that the incoming g-code would command a change of direction just after a step - see: https://github.com/KevinOConnor/klipper/issues/196#issuecomment-416237533

-Kevin