Klipper3d/klipper #1159: When BL-Touch is next to bed on startup - the printer doesn't start

dushyantahuja, 2019-01-26:
Hi,
After the latest updates to Klipper, BLTouch is working fine, except when:

1. The printer was switched off with the bed right next to the nozzle
2. When you switch on the printer, Klipper stays disconnected
3. The relevant entries in the log are:

Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 137, in _connect
    cb()
  File "/home/pi/klipper/klippy/extras/bltouch.py", line 72, in handle_connect
    raise self.printer.config_error(str(e))
Error: BLTouch failed to raise probe

The only way I could get the printer to work was to remove the BLTouch section from the config file, switch on the printer and force it to move up, and then re-instate the BLTouch section.

Suggested resolution:
1. The printer connects and provides a message in the console (not only in the log)
2. The printer comes up in force move (warning the user to be careful to move)

[klippy.log](https://github.com/KevinOConnor/klipper/files/2799442/klippy.log)

fredizzimo, 2019-01-27:
I reported this twice before the new version was merged, once when doing the code review, and once when I did this test https://github.com/KevinOConnor/klipper/pull/1091#issuecomment-453749542. The error message is also misleading, because it's not really trying to move the pin up.

An easier workaround, at least if you can mechnically move the z-axis, for example by turning the stepper by hand, is to turn off the power of the printer, then move the z-axis.

IMO, the correct way to handle this, is the way I do in my pull request https://github.com/KevinOConnor/klipper/pull/982, is to try to reset the probe, and pull the pin up. And only if that fails, report the error. The reason why it fails originally is that the probe does a self-test on startup, and that can't be performed too close to the bed. Because it's not a real error, I think it's pretty safe to do so. If there are some other errors, like the pin can't move up, or bad wiring the error is generated.

Handling it that way, does not require any user action, and everything will work as it should. There should also be no risk of damaging the probe, because the pin is always moved up before the printer is started.

KevinOConnor, 2019-01-28:
As mentioned, this was reported earlier in PR #1091.  It is also my understanding that the bltouch will fail its self test if the bltouch is too close to the bed.  The current code does issue a "reset" and "pin_up" request during startup.  We can issue some other sequence (or issue this sequence multiple times) if that will help.  It's not clear to me what command sequence will suffice.

I'm told that it is difficult to clear errors from the bltouch (and especially the clones).  In that case, it might make sense to change your end gcode and cancel gcode scripts to move the head to a position such that the bltouch can pass its self test.

-Kevin

jakep82, 2019-01-29:
I ran into the same issue as the OP over the weekend.  I didn't keep the log, but I had the exact same situation.  My nozzle was too close during startup which caused the BLTouch self-test to fail (genuine BLTouch).  I was then unable to connect to the printer until the error state on the BLTouch was cleared, but I couldn't clear it without connecting to the printer.  I ultimately moved the bed down by hand and power cycled the controller board.  The BLTouch then successfully completed the self-test and I was able to connect to the printer.

jakep82, 2019-01-30:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2811039/klippy.log)

Log from testing tonight.

dushyantahuja, 2019-01-30:
As far as I know, that the error will not get reset until the pin can move freely. Is it possible to not disconnect the printer on this issue and simply ignore it (or simply warn the user)?

The user cannot home or move the printer at this time - as the error will come up when trying to home; however it is still possible to force move the printer using SET_KINEMATIC_POSITION X=100 Y=100 Z=10 and then moving the Z axis up. Once the pint is free - the BL-Touch can be reset and used to home.

Hence, my recommendation would be to warn on the error and not disconnect.

As an aside, I have a belted Z with a worm gear hidden at the bottom of my printer - so nearly impossible to manually move the Z without breaking anything. Most of the time the printer moves down before stopping, but sometimes - I need to switch off when I've done something wrong - and that kills it.

KevinOConnor, 2019-01-30:
@dushyantahuja - I made two changes to the work-bltouch-debug-20181211 branch yesterday: retry the reset/pin_up sequence on an error, and continue with the startup procedure even if the code can not confirm the reset/pin_up succeeds.

cd ~/klipper ; git fetch ; git checkout origin/work-bltouch-debug-20181211 ; sudo service klipper restart

@jakep82 - I think that branch should now have a fix for the issue you were seeing.

-Kevin

fredizzimo, 2019-01-30:
@dushyantahuja,

I believe you are wrong for the vast majority of the probes out there. I know for the fact that a reset is enough on my Genuine BLTouch Smart. I'm also fairly sure it was enough for @danielkucera with his clone, which I believe is a "Based On Anctlabs" TL Touch. The reason why I think that's the case for him, is  because we did some tests together, which should have showed some problems if there are any. A reset should also be enough on a 3D touch according to this https://github.com/MarlinFirmware/Marlin/issues/5817#issuecomment-280989045

Note that the special case where it doesn't work on a 3D Touch is when the pin is prevented from moving down at all. This can't really happen when the probe is mounted normally, unless it's touching an old print or something, but that requires manual invention anyway. Or alternatively when there has been some real problem with the probe, in which case I don't think it's wise to continue. In any case the problem is fixed when the printer is shutdown, and the obstacle removed, after which a normal reset should be enough if it's still too close to the bed. The Marlin firmware compensates for this error by first moving the pin down then up, after reset. However that can't be done when we are already too close to the bed, which is the case here. I also think it's generally is a bad idea to try to command the pin down after an error. 

That post also discuss some kind of "new" probe, which indeed might have some problems. However, from checking the Marlin code and also reading the issue, it looks like Marlin decided to not support that probe at all. It's probably very rare, and not sold anymore, or if it is, then it's sold with the issue fixed. Therefore I think we can ignore support for that probe.

Note that reset alone, might not pull the pin up, so the correct sequence is
1. Reset
2. Pin Up
3. Wait for enough time (at least one second), this is `pin_move_time` in the configuration. Internally the probe tries to move the pin up for some time until it triggers a failure. This time is considerably longer than the time it takes for the pin to move up in ideal cases.
4. Check that the error has cleared

@KevinOConnor, I suggest that you remove the log and continue part of the code, since as I explained above, I don't think it's a problem on any of the common probes in circulation. Or at least that you make it an option that is disabled by default. The reason for that is, because it's IMO, a very bad think to leave the pin down, touching the bed. Any sideways movement at this point, could destroy it. 

Imagine the following case, which I think is fairly probable. Someone just bought a new BLTouch, but connected it wrongly, so that it enters the error state upon startup. The pin was left down touching the bed, because that's how the user left it, when connecting it at the right height. The user also forgot to configure a retract distance when homing, so now when he runs G28, the head moves to the side, and destroys the probe before he even had a chance to try it at all. Or maybe even worse, the probe accuracy got really bad, causing a lot of frustrating in addition to the money lost.

I also still don't think that relying on the endstop code to do the timing is working correctly. Which in this case could mean that the code thinks that the error condition is clear when it in fact shouldn't be. Which means that it could get past the error, even if the pin failed to move up. I have to do a proper repeatable test with your code to confirm it. But from the tests I did with my code, I could not get it to work, so there's a reason why I'm relying on wait moves to make the error detection robust.

Finally one thing that can't be repeated enough: **Everyone that hasn't changed the default `pin_move_time` to 1 or more, do it so now!** (I can see that @jakep82 hasn't set it) The default 0.2 seconds is way too low, and will almost never catch the error in time. The Marlin guys measured the time that it takes to 0.65 seconds, I determined that 0.5 seconds can sometimes detect the errors, but in many cases it failed. The Marlin firmware itself waits for 1.5 seconds for the error (regardless of what you have the `BLTOUCH_DELAY` set to) But really @KevinOConnor, this has to be changed by default.

dushyantahuja, 2019-01-30:
> @dushyantahuja - I made two changes to the work-bltouch-debug-20181211 branch yesterday: retry the reset/pin_up sequence on an error, and continue with the startup procedure even if the code can not [...]

@KevinOConnor - this works for me. After I switched off the printer with the bed right next to the nozzle, I received the following warnings on the printer being switched on:

Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // Failed to verify BLTouch probe is raised; retrying.

And then Klipper connected to the printer - which is good. I was then able to issue a G28 command (my default setting moves the bed down by 5mm regardless). The bed went down, but BL-touch still did not reset and I got the following errors:

Recv: !! BLTouch failed to deploy
Recv: !! BLTouch failed to deploy

However issuing the following commands reset the BL-Touch

BLTOUCH_DEBUG COMMAND=reset
BLTOUCH_DEBUG COMMAND=pin_down

and everything worked after that. 

I'm happy with this - at least I was able to recover from the error without having to move the Z axis manually.
Thanks a lot.

dushyantahuja, 2019-02-01:
Great - Thanks

closed 2019-02-01 by dushyantahuja

fredizzimo, 2019-02-17:
@KevinOConnor,

These changes do not work! It does **not** clear the error state when starting the printer, even on a genuine BLTouch smart. While it's true that you can use some debug commands, to get rid of it. I don't think it's a good option.

I tested both with the master and the debug branch.

And please don't tell me that it's impossible, because my code clears the error just fine and automatically. 

In Marlin the error checking and clearing can be seen here
https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L2087

Also, as you can see there's a 1500ms delay, not 675ms, for doing that. Here's the place where Marlin resets the error at startup https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/Marlin_main.cpp#L15232. And yes, they call `stop()` when it fails, as can be seen in the `set_bltouch_deployed` function.

I also don't see where you get 675 from, Marlin sets  `BLTOUCH_DELAY` to 375 (not 675) here https://github.com/MarlinFirmware/Marlin/blob/85c82ab28e13cb25043ae166062785bff39434b9/Marlin/Conditionals_LCD.h#L476. But in Marlin that delay is unrelated to the error checking delay, so we can't use that value as a baseline. Also according to my tests, 500ms was too low, some errors got undetected, and 675 is awfully close to that, so I'm not sure if it's safe or not. But considering the small margin and the amount of different probes, I would bet it's not.

Finally, I don't know why you keep talking about regressions. These are just plain old bugs, which need to be fixed, neither the previous version or this version work as they should. In one case, errors are completely ignored, in the other case, it fails to start. From my point of view they could potentially be called regressions though, since my code does not have these problems.