Klipper3d/klipper #1163: The new BLTouch code has several problems

fredizzimo, 2019-01-27:
I'm resubmitting this bug report from the pull request, so that the bugs don't get lost. These tests were done with a genuine BLTouch Smart sensor. 

I'm starting `pin_move_time` set to `1`

Run `G28`, then `PROBE`. It correctly fails to probe, because the pin is too close to the bed. However the error message reported is "BLTouch failed to raise probe", or **exactly the opposite** of what it's supposed to be: "BLTouch failed to deploy the probe"

But that's not all, **the pin is left down**, and the probe is in an error state. This is in itself not a super critical issue. But the following is **the steppers are not disabled!**. This means that if you perform any movement, perhaps as a part of a script that you have no control of, then **it starts to move with the pin touching the bed**. This can very likely **destroy the probe**, becuase it's a very sensitive device. That's a complete showstopper. 

IMO, if the BLTouch is in an unrecoverable error state after a probing attempt, it should shut down the printer to prevent further problems. This case is not really unrecoverable on most probes though, since you can reset it and move the pin up to get rid of the error. Some clones don't respond well to that, so on those a shutdown might be necessary in order to avoid damage.

Let's then continue to the bug immediately following the previous steps. Issue a `restart`. The **printer fails to restart** with the following error

```
Recv: // BLTouch failed to raise probe
Recv: // Once the underlying issue is corrected, use the "RESTART"
Recv: // command to reload the config and restart the host software.
Recv: // Printer is halted
Recv: // Klipper state: Not ready
Recv: !! BLTouch failed to raise probe
```
The only way to get rid of this error is a cold restart. Or is it? **Unfortunately it isn't**. If you didn't issue a move command when you got the error, the probe is still too close to the bed, and the self-test will fail. So you get the same error message again. **This time the error message is wrong again**, notice how the pin is up, while the error claims that it failed to raise the probe.

So how can you get rid of this error? You have to shut down the printer, move the z-axis by hand, then restart it. This will confuse many users, especially since it's very easy to repeat, Issue a `G28` then shut down the printer without doing anything else.  **Notice how the printer fails to restart**. There's also no inidication of what might be wrong, and the error message is completely misleading.

Now let's do something dangerous, set `pin_move_time` to the default `0.2`, **which is way too low** and useless, as I will demonstrate.

Issue a `G28`, then `G1 Z0.0`, then `PROBE`. Observe how the printhead moves down, and **crashes into the bed**. Again, it wrongly reports that it failed to raise the probe.

Notice that just like I predicted during the review the following code fails to catch the issue 
```python
for s, mcu_pos in self.start_mcu_pos:
    toolhead.dwell(self.pin_move_time + MIN_CMD_TIME)
        if s.get_mcu_position() == mcu_pos:
            raise homing.EndstopError("BLTouch failed to deploy")
```
And also that even when the `pin_move_time` is set high enough, it still performs a small amount of movement, and that test fails to detect the issue.

For a reference, my pull request https://github.com/KevinOConnor/klipper/pull/982, handles all these cases correctly.

KevinOConnor, 2019-02-08:
It's not really clear to me what is being reported here.  I don't feel the error message is that bad - though we can certainly change the wording.  As of commit 345a5f4e the default pin_move_time is 675ms (which matches Marlin).  As for error tests - from my quick review, I don't see that Klipper is worse than any other firmware.  (Indeed, as near as I can tell, Klipper is dramatically better at error checking.)  That doesn't mean Klipper shouldn't strive to be better, but additional error checks will need to be balanced with usability (works with clones, doesn't increase probing time for the normal case, etc.).

-Kevin

arminth, 2019-02-10:
@KevinOConnor 
I don't think it's really about wording of the error messages (whereas the remark of @fredizzimo  is IMHO correct).

What is indeed very clear here is what is reported regarding issuing a PROBE after G28. Exactly this is happening! You can't get the printer out of this error condition without shutting downand moving by hand. I have tested it with an original BLTouch as I got rid of my crappy clones.
I didn't want to reproduce the nozzle crash though! ;)

I know that the both of you have pretty diverging opinions but I think on these obvious issues, for the sake of nozzles, printbeds and BL-touch probingpins, cooperation might be more valuable than confrontation.

Jm2€cts.

Cheers
Armin

KevinOConnor, 2019-02-12:
@arminth - it isn't clear to me what you are reporting.  I suspect you are reporting that the bltouch goes into an error state if one issues a PROBE command when the bltouch is close to the bed.  If so, this is a "feature" of the bltouch.  The error can be cleared in software - there was a discussion on this at: https://github.com/KevinOConnor/klipper/issues/1159#issuecomment-459118130 .  However, given the reports that some clone's can't easily clear an error state, I'd recommend not issuing a PROBE command when the head is near the bed.

If you're reporting something else, I recommend you open a new github issue and include the klipper log file from the event - as described at: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

fredizzimo, 2019-02-17:
What's reported here:

1. When you do a `PROBE` command starting too close to the bed:
* The error message is wrong, it says it fails to raise the probe, when in fact it fails to put it down!
* The probe is lowered, despite the message. 
* The steppers are left enabled, which means that you can now do movement commands and destroy the probe
* I would like to see the printer shutdown, but only if resetting fails to work (like in my code). Which in practice means that the printer remains in a working state with the probe in it's retracted state when it's too close to the bed. But if there's a hardware issue, the printer is shutdown.
2. The startup issues, which are reported here https://github.com/KevinOConnor/klipper/issues/1159 and not fixed. 
3. The pin move time, too low, and it still is with 675, see my comments in the other issue
4. I think, but I'm not 100% certain, that there could be a small movement in case 1., even when the `pin_move_time` is set high enough. But I think my conclusion from my tests above are wrong. I read the code and my report again, and I no longer come to the same result. Maybe the code is changed, or I didn't check it carefully enough when writing that.

Just like @arminth, I'm too wondering why you are so resistant to accepting that your code has bugs, which in this case are quite fatal. Virtually all code, has bugs, so that's nothing to to be ashamed of. Actually I'm even surprised it has so few, because you don't even have the hardware to test it on. I'm not claiming that my code is perfect in any way, bug it doesn't have known clearly demonstratable bugs like these. It's of course possible that it has, but no one has pointed them out to me yet.

I suspect the real reason, for not accepting, neither my bug reports or my code, or even to give a real constructive review of my code, so that I know what needs to be changed, is my behaviour. I know I am and have been very direct, bordering to aggressive, but that should not be taken personally. And if you have, I appologize for that. All my intent has been to have a reasonably bug free implementation of BLTouch in Klipper (written by you, me or someone else), and I feel being direct to the point, is the most effective way to get there. Clearly that hasn't been the case, since I have spent way, way more time writing these kinds of messages, than actually implementing my stuff, and I'm sure it's the same way for you to.

How can this process be improved? I see two ways
1. You promptly respond to my bug reports, first by acknowledging the issues, so I know that I don't have to explain them again. Then you actually fix them all in the next version. 
2. You tell me exactly why my pull request is not good, and I fix the issues myself. Or alternatively if it's something that is easier to do in practice, modify my pull request with the changes you want, let me know that so I can verity and test that no regressions were introduced by doing that.

What I particularly don't like, is to accept that there are bugs, but not fix them, especially when fixing them is completely viable and a matter of just a couple of hours programming at most.

KevinOConnor, 2019-02-25:
I'm not interested in reading or responding to messages like the above.  It is important to me that all communication occur in a neutral tone and that the base assumption is that both parties are making a good faith effort to improve the software.

Unfortunately, my conclusion is that you and I are not going to work well together.  I don't think it is useful for either of us to continue to invest time pretending otherwise.

Best of luck,
-Kevin

closed 2019-03-04 by KevinOConnor