Klipper3d/klipper #1165: [controller_fan] is not a valid section

itsthejoker, 2019-01-28:
Hi there! When applying the config found here https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L432-L454 in an attempt to get automatic control of a LED light bar with higher PWM requirements than a fan, I get the error that [controller_fan] is not a valid section. Log is included.

Cheers!

[klippy.log](https://github.com/KevinOConnor/klipper/files/2801010/klippy.log)

pecirep, 2019-01-28:
Looks like you're missing a mandatory option. Did you define a heater this fan associated with?

itsthejoker, 2019-01-28:
Default is "extruder", which matches my config. I'll throw it in just to verify, though.

pecirep, 2019-01-28:
can you post the entire section?

itsthejoker, 2019-01-28:
Here's the entire [controller_fan] section with all flags set. It continues to give the same error as above.

```ini
[controller_fan]
pin: ar3
cycle_time: 0.005
hardware_pwm: False
max_power: 1
shutdown_speed: 1
kick_start_time: 0.100
idle_timeout: 60
idle_speed: 1
heater: extruder
```

pecirep, 2019-01-28:
I've just copy-pasted that section to my printer and it works. This is a somewhat recent addition, have you updated klipper?

itsthejoker, 2019-01-28:
That... is a great question. Let me take a look at that.

itsthejoker, 2019-01-28:
Shit, that was the answer. I should have checked that. Thanks, mate!

closed 2019-01-28 by itsthejoker

pecirep, 2019-01-28:
oh yeah, I killed that recently, thanks for the heads-up 😄