Klipper3d/klipper #1166: adjust axis movement when direction change

rizpardazan, 2019-01-28:
hello all,
my z axis have a lead screw and a gearbox , when i move axis 1mm up it moves exactly 1mm but after that when i move 1mm down printer shows it is in 1mm down but  real movement is 0.95 to down because of gearbox internal leaping.
every time i change direction from up to down or down to up i have this 0.05 lost movement,
how can i adjust this in klipper to add 0.05mm to axis movement when direction changes?

Hywelmartin, 2019-01-28:
It has been discussed before in  #1144 for example..
On way to work around this back lash is to always move upwards.. for example to always move down 1 mm right before layer change and then up to the desired position. in this case the vibrations and the weight of the gantry wont suddenly make the z to sink..

[image]

rizpardazan, 2019-01-28:
> It has been discussed before in #1144 for example..
> On way to work around this back lash is to always move upwards.. for example to always move down 1 mm right before layer change and then up to the [...]

it is not so important in layer change, it is important when we use auto bed level and when printing z axis moves up and down to keep layer height but without backlash correction auto bed level can not do the job.

do you know where is the movement function in the source?

FHeilmann, 2019-01-29:
It is really in your best interest to fix your printer instead of looking to software to fix the mechanical issue.

rizpardazan, 2019-01-29:
> It is really in your best interest to fix your printer instead of looking to software to fix the mechanical issue.

Backlash setting is an advanced option in all professional cnc controllers and make it easy to resolve many problems,
I do my best but backlash is in the essence of the system and never be zero except with backlash setting.

KevinOConnor, 2019-01-29:
Adding "backlash" support would require code changes and a developer ready to implement it.  (It's not something I, personally, will have time to implement.)  As for where to look in the source code, see the docs/Code_Overview.md file.

-Kevin

rizpardazan, 2019-01-31:
> Adding "backlash" support would require code changes and a developer ready to implement it. (It's not something I, personally, will have time to implement.) As for where to look in the source code, see the docs/Code_Overview.md file.
> 
> -Kevin
Thank you kevin ,
I think adding Backlash no need major changes and did not hard for you to do that , because of that you write this firmware and know that where should add codes , you should only add a variable for X and Y and Z axis and add this variable value to axis movement when direction changes.

mzbotreprap, 2019-01-31:
> I think adding Backlash no need major changes and did not hard for you to do that , because of that you write this firmware and know that where should add codes

Please don't confuse knowledge of the code base with ease of software feature implementation. "How hard can it be?" is usually followed by lessons in humility.

I'm not implying this would not be useful, especially on a Klipper powered CNC, but it's not an easy feature and will require development time which Kevin doesn't have to spare at the moment.

KevinOConnor, 2019-02-18:
I'm going to mark this issue as inactive as it doesn't look like anyone is currently working on it.

-Kevin

closed 2019-02-18 by KevinOConnor