Klipper3d/klipper #1167: Request - Set / Adjust PID parameters via gcode

bruce356, 2019-01-28:
Possibility for PID values be changed for different temperatures / materials using gcode macros?
In Marlin one can use in the slicer start script:-
M304 - Set Bed PID
M301 - Set Hotend PID
to change preset pids
thanks - bruce

KevinOConnor, 2019-01-30:
I'm sure it's possible, but someone would need to do the development work to implement it.

-Kevin

KevinOConnor, 2019-02-18:
I'm going to mark this as inactive as it doesn't look like it is currently being worked on.

-Kevin

closed 2019-02-18 by KevinOConnor

bjones14, 2020-01-02:
This is something I could potentially work on...

I think a good implementation would be to create a Klipper macro (not sure if that's what they are called?  not a G or M code, but something like PROBE or PROBE_CALIBRATE) that would allow setting PID values for a specified heater in the configuration file.

This macro would need to reset the PID so that the integral is zeroed out and then restart the control loop with the new PID values specified from the command.

I would assume there is a mechanism for this to happen without requiring a restart of the firmware @KevinOConnor?  I don't think needing to reload the configuration file is feasible but I am unfamiliar with the code structure.