Klipper3d/klipper #1170: Saving Z-offset with manual bed mesh (G29 S4 Zn / M851)?

JohnEdwa, 2019-01-29:
TL;DR version: when using (manual) mesh bed leveling, there should be a way to edit and save the Z offset to the configuration from the LCD and G-code, which is added to the mesh points when printing, replicating the Marlin G29 S4 Zn and M851 commands.

If there already is, I didn't find it.  
What is the current way to set the manual bed mesh offset in Klipper, start G-code in Octoprint? 

---

In Marlin, after doing [the Manual Mesh Leveling (G29)](http://marlinfw.org/docs/gcode/G029-mbl.html), you would use the LCD to set the offset to match whatever you used as the gauge to level the bed (thickness of the paper for example) - the G-code it used for this was `G29 S4 Zn`:
> S4: Set a global Z offset. Positive values are away from the bed; negative values are closer.  

It would then get saved with the mesh as a global offset added to all the mesh points at runtime to actually make them match the bed and not the bed+paper thickness, done this way as your gauge isn't going to always be the same thickness so it can't be a hardcoded value.
During the print, the [Tune menu has an item called "Bed Z:"](http://marlinfw.org/docs/features/lcd_menu.html#tune) which then can be used to edit this value in real time to achieve the perfect first layer (and it is saved automatically) - it can basically replace babystepping whenever a mesh (manual or probed) is loaded and being used.

In Klipper, `SET_GCODE_OFFSET Z` does the height adjustment, but this value isn't saved to the configuration (although the mesh saves x_offset and y_offset?) and the only ` z_offset` I could find in the printer config are with a `[probe]` or `[bltouch]`, although those would be changed in Marlin using [M851](http://marlinfw.org/docs/gcode/M851.html) anyway.

KevinOConnor, 2019-01-30:
I agree that it should be possible to specify the "probe" offset when performing manual probing.  My high level thinking was to add it as an additional parameter to the XXX_CALIBRATE commands (eg, DELTA_CALIBRATE METHOD=manual MANUAL_PROBE_OFFSET=0.01 ).

-Kevin

JohnEdwa, 2019-01-30:
It would match the `G29 S4 Zn`, though how would Klipper react is someone fumbled a "BED_MESH_CALIBRATE" in the middle of a print, would it just start doing it as asked?

Adding `SET_ZPROBE_OFFSET X=n Y=n Z=n` would be the most obvious solution, though another that came to mind is `BED_MESH_OUTPUT Z_OFFSET=0.01`, logic being it's an offset used only with those values and it is saved into the mesh data. And the worst you can do is print the mesh data.

KevinOConnor, 2019-01-31:
I'm not sure what you are suggesting.  I was not suggesting that it should be possible to change the offset after a calibration routine - I was suggesting that it should be possible to specify the manual probe offset when doing the calibration itself.  If there is a desire to change the offset at some future point then SET_GCODE_OFFSET is possible.  Otherwise, one can manually alter the config by changing the position_endstop settings. Or, one can rerun the calibration routine.

-Kevin

KevinOConnor, 2019-02-03:
FYI, I define a PRINT_START gcode_macro in my printer config and I set my slicer to issue that gcode macro at the start of prints.  If I need to modify something like the z height, then I just change that macro in my config and issue a RESTART.

It's not clear to me what you are suggesting, but it sounds like it would require software development.  So, it will require a developer to implement and then submit a pull request.

-Kevin

JohnEdwa, 2019-02-03:
And I currently do it from my octoprint start Gcode, as having to connect to the Pi, edit a config file and then issue a reboot is a rather annoying hassle, which is why this issue exists.

But hmmh, I guess Icould try to have a go at it myself as well. 
At it's simplest it's a single input Gcode command saving a Z-offset to the config bed mesh table that is added to the data. "SET_MESH_OFFSET z=0.05" saving "[bed_mesh] z_offset = 0.05" which turns:

```
       0      1       2
0 +0.011 -0.020  -0.026
1 +0.017 +0.002  -0.019
2 +0.022 -0.030  -0.013
```

into

```
z_offset = 0.05
       0      1       2
0 +0.061 -0.030  -0.036
1 +0.067 +0.052  -0.049
2 +0.072 -0.020  -0.043
```

...how hard can it be?

drphil3d, 2019-02-06:
What I do is use the SET_GCODE_OFFSET Z_ADJUST=.05 to get the nozzle to the desired height on the first layer. Then after the test print is finished I use GET_POSITION to get the gcode offset and then apply that to my probe offset. It would be so much easier if there was a command like what marlin has that automatically updates the eeprom with the adjustment.
M290 http://marlinfw.org/docs/gcode/M290.html

KevinOConnor, 2019-02-14:
FYI, several helper utilities were merged recently (including PROBE_CALIBRATE) - commit 57e6acbc.

-Kevin

KevinOConnor, 2019-03-04:
I'm going to close this.  For now the recommendation is to set this in a START_PRINT gcode macro.

-Kevin

closed 2019-03-04 by KevinOConnor