Klipper3d/klipper #1171: Extended G-Code for max_extrude_only accel/velocity?

ewb-git, 2019-01-29:
Hello,

I was looking at the extended gcode commands, and I saw you have a lot of useful commands for speeds and velocities, which would allow for easy testing with a "if layer z" command in many slicers.

I was also interested in testing the max_extrude_only accelerations and velocities, however, I believe there aren't any specific extended gcodes to target these? I might be wrong since I'm new to Klipper, but if they don't exist I think they could be useful for calibration/testing purposes.

Thank you!

KevinOConnor, 2019-01-30:
There are currently no commands to change the "extrude_only" config settings at runtime.  Though it is possible to alter the settings in the file and then issue a RESTART command.

-Kevin

ewb-git, 2019-01-30:
Yea I was thinking for tuning extruder acceleration/velocity, it would be nice to have an option to print a multi-level speed tower using z height. That way you can automate everything into 1 or 2 prints instead of having dozens of small print operations that need to be labeled, time consuming, and difficult to compare.

Well, it would be a nice feature to consider. Thanks!

KevinOConnor, 2019-02-18:
I'm going to mark this as inactive as it doesn't look like it is currently being worked on.

-Kevin

closed 2019-02-18 by KevinOConnor