Klipper3d/klipper #1174: BlTouch not probing when i Delta calibrate

lightface79, 2019-01-30:


lightface79, 2019-01-30:
Hi!!!   
Got my bltouch working on my delta =) 
got it mounted  staight under the nozzel
so offsets x & Y 0    Z 50

i can run a g28 & Probe and it probs the Z height.
All is well.

but i cant run a g28/home  and then a Delta_calibrate
it just homes and thats it. it dont deploy the bltouch pin
or move down. 
it just stays in the Top Home possisitinon.

[printer.txt](https://github.com/KevinOConnor/klipper/files/2811958/printer.txt)

lightface79, 2019-01-30:
tried one more time..  
with some ice im my tummy.
after it lowers Z it stops.
then i get thees errorors in my terminal

Send: G28
Recv: ok
[...]
Send: delta_calibrate
[...]
Recv: !! BLTouch sensor test failed
Recv: !! BLTouch sensor test failed
Recv: ok

KevinOConnor, 2019-01-31:
In order for us to assist, you'll need to attach the Klipper log file - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

-Kevin

daniHagl, 2019-02-08:
I don't have a BLTouch probe, but still have the same behavior. If I issue the "DELTA_CALIBRATE METHOD=auto" command, the toolhead moves to the spezified position of z=20 and stops. No Errors in the terminal.

[klippy.zip](https://github.com/KevinOConnor/klipper/files/2845463/klippy.zip)

KevinOConnor, 2019-02-08:
@daniHagl - the command "DELTA_CALIBRATE METHOD=auto" is not valid.  If you wish to perform a calibration with an automatic bed sensor, issue just a "DELTA_CALIBRATE" command.

closed 2019-02-08 by KevinOConnor