Klipper3d/klipper #1175: Question for Pressure Advance

AndyWo83, 2019-01-30:
Hello,
must I disable the retract in my slicer settings to find the correct parameters? 
In the documentation is nothing written about this. But I think the retract needs to be disabled for this test?
Thanks

AndyWo83, 2019-01-30:
I have in all my slicer settings a "retract on layer change".
In my opinion this would change the result, or you have to ignore this position.

numanair, 2019-01-30:
Without retraction you might get one extra-blobby corner. Either way you should ignore this corner.

KevinOConnor, 2019-01-31:
As mentioned above, the pressure advance test described at https://github.com/KevinOConnor/klipper/blob/master/docs/Pressure_Advance.md is robust to a wide variety of typical slicer settings.  So, print with "typical defaults".  Normal retraction settings should not impact the test, but if you are concerned then disable retraction.

-Kevin

closed 2019-02-04 by AndyWo83