Klipper3d/klipper #1180: Klipper MMU Based on RAMPS

SergeyKrutyko, 2019-02-01:
Hello Kevin, i am back :) Remember topic start #700 

My Project of Custom MMU is alive: http://3dtoday.ru/upload/main/ce3/ce3115ebdb8e4b0cb2e996e0399989ea.jpg

I need your help - I could not find another way to contact you (there is no your email)

I write first config for MMU:
`# ORIGINAL CONFIG:
[extruder]
step_pin: ar26
dir_pin: ar28
enable_pin: !ar24
step_distance: .002351646
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar10
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog13
control: pid
pid_Kp: 31.315 
pid_Ki: 1.725 
pid_Kd: 142.091
min_temp: 10
max_temp: 350
max_extrude_only_distance: 350.0
pressure_advance: 0.20
pressure_advance_lookahead_time: 0.010
max_extrude_cross_section: 65
#----- new entry 5.12.18 for retract
max_extrude_only_velocity: 100.0
max_extrude_only_accel: 6000.0
#additional
activate_gcode:
    SET_PIN PIN=mmu_led1 VALUE=.5
    MMU_SWITCH_FILAMENT 0
    SET_PIN PIN=mmu_led1 VALUE=1

deactivate_gcode:
    MMU_UNLOAD_FILAMENT
    SET_PIN PIN=mmu_led1 VALUE=0

# ADDITIONAL MMU CONFIG ====================================================================
[mcu mmu]
serial: /dev/ttyUSB1
pin_map: arduino

[extruder1]
step_pin: ar26
dir_pin: ar28
enable_pin: !ar24
step_distance: .002351646
nozzle_diameter: 0.400
filament_diameter: 1.750
shared_heater: extruder
min_temp: 10
max_temp: 350
max_extrude_only_distance: 350.0
pressure_advance: 0.20
pressure_advance_lookahead_time: 0.010
max_extrude_cross_section: 65
max_extrude_only_velocity: 100.0
max_extrude_only_accel: 6000.0

activate_gcode:
    SET_PIN PIN=mmu_led2 VALUE=.5
    MMU_SWITCH_FILAMENT 1
    SET_PIN PIN=mmu_led2 VALUE=1

deactivate_gcode:
    MMU_UNLOAD_FILAMENT
    SET_PIN PIN=mmu_led2 VALUE=0

[extruder2]
step_pin: ar26
dir_pin: ar28
enable_pin: !ar24
step_distance: .002351646
nozzle_diameter: 0.400
filament_diameter: 1.750
shared_heater: extruder
min_temp: 10
max_temp: 350
max_extrude_only_distance: 350.0
pressure_advance: 0.20
pressure_advance_lookahead_time: 0.010
max_extrude_cross_section: 65
max_extrude_only_velocity: 100.0
max_extrude_only_accel: 6000.0

activate_gcode:
    SET_PIN PIN=mmu_led3 VALUE=.5
    MMU_SWITCH_FILAMENT 2
    SET_PIN PIN=mmu_led3 VALUE=1

deactivate_gcode:
    MMU_UNLOAD_FILAMENT
    SET_PIN PIN=mmu_led3 VALUE=0

[extruder3]
step_pin: ar26
dir_pin: ar28
enable_pin: !ar24
step_distance: .002351646
nozzle_diameter: 0.400
filament_diameter: 1.750
shared_heater: extruder
min_temp: 10
max_temp: 350
max_extrude_only_distance: 350.0
pressure_advance: 0.20
pressure_advance_lookahead_time: 0.010
max_extrude_cross_section: 65
max_extrude_only_velocity: 100.0
max_extrude_only_accel: 6000.0

activate_gcode:
    SET_PIN PIN=mmu_led4 VALUE=.5
    MMU_SWITCH_FILAMENT 3
    SET_PIN PIN=mmu_led4 VALUE=1

deactivate_gcode:
    MMU_UNLOAD_FILAMENT
    SET_PIN PIN=mmu_led4 VALUE=0

[extruder4]
step_pin: ar26
dir_pin: ar28
enable_pin: !ar24
step_distance: .002351646
nozzle_diameter: 0.400
filament_diameter: 1.750
shared_heater: extruder
min_temp: 10
max_temp: 350
max_extrude_only_distance: 350.0
pressure_advance: 0.20
pressure_advance_lookahead_time: 0.010
max_extrude_cross_section: 65
max_extrude_only_velocity: 100.0
max_extrude_only_accel: 6000.0

activate_gcode:
    SET_PIN PIN=mmu_led5 VALUE=.5
    MMU_SWITCH_FILAMENT 4
    SET_PIN PIN=mmu_led5 VALUE=1

deactivate_gcode:
    MMU_UNLOAD_FILAMENT
    SET_PIN PIN=mmu_led5 VALUE=0

[stepper_mmu_sel_filament]
#use stepper X on RAMPS
step_pin: mmu:ar54
dir_pin: mmu:ar55
enable_pin: !mmu:ar38
# for standart stepper nema17 and divider 1/16 with 8mm/2mm leadscrew = 1600 steps/mm or 1/1600
step_distance: .000625
endstop_pin: ^mmu:ar3
#offset from endstop to last filament pos in mm
position_endstop: 5
position_max: 150
homing_speed: 50

[stepper_mmu_select_extruder]
#use stepper Y on RAMPS
step_pin: mmu:ar60
dir_pin: !mmu:ar61
enable_pin: !mmu:ar56

#distant in grade 360 full loop * 16 / 200 = 8.888888 (1/8.888888)
step_distance: .1125

[stepper_mmu2_load_filament]
#use stepper E0 on RAMPS

step_pin: mmu:ar26
dir_pin: mmu:ar28
enable_pin: !mmu:ar24

#MK8 Effective diameter: 6.7mm circumference : 21mm
#Steps per mm: 152 approx. - tested with direct drive, 200 steps/rev and 1/16th. microstepping: 200 * 16 / 21 ~= 152 (1/152)
step_distance: .0065789473684211

[mmu]
#distance from hotend to extruder in mm
hotend_len: 50
#distance from extruder to mmu FINDA in mm
extruder_len: 300
#distance from FINDA to parking position in mm
parking_len: 40
#min temperature to chage filament
change_temp: 200
#distance X between filaments in mm (offset position setup in stepper_mmu_sel_filament as position_endstop)
dist_filament: 15

#buttons to manual select at AUX4 (pullup)
mmu_but1: !mmu:ar32
mmu_but2: !mmu:ar47
mmu_but3: !mmu:ar45
mmu_but4: !mmu:ar43
mmu_but5: !mmu:ar41

#leds for show selected at AUX4
[output_pin mmu_led1]
pin: mmu:ar39
pwm: True
#low speed for setup blinking mode if it need
cycle_time: 0.500

[output_pin mmu_led2]
pin: mmu:ar37
pwm: True
cycle_time: 0.500

[output_pin mmu_led3]
pin: mmu:ar35
pwm: True
cycle_time: 0.500

[output_pin mmu_led4]
pin: mmu:ar33
pwm: True
cycle_time: 0.500

[output_pin mmu_led5]
pin: mmu:ar31
pwm: True
cycle_time: 0.500

#END CONFIG`

SergeyKrutyko, 2019-02-01:
i have few questions:

1. How i can get autohomming of stepper_mmu_sel_filament, may be i must use: G28 mmu_sel_filament ? If this is not the case, then tell me how to implement it correctly?
2. how do i get the current temperature of the hotend from the plugin? My task is to get the temperature of hot-end, and if it is below the allowable temperature, then before changing the plastic, first heat it up (for example, by doing the GCODE or perhaps you can tell me how to do it correctly)?

Thanks!

SergeyKrutyko, 2019-02-01:
3. i have one sensor connected to endstop Y or RAMPS. In some cases need move stepper_mmu2_load_filament until the sensor switch. How i can made it correctly?

SergeyKrutyko, 2019-02-01:
4. In the system will be 5 extruders with a virtual one heater. If it possible to also make virtual extruders in order not to list these parameters:
.....
step_pin: ar26
dir_pin: ar28
enable_pin:! ar24
step_distance: .002351646
.....

KevinOConnor, 2019-02-03:
There was a long discussion on something similar with @trevjonez - see #916, #1044, #1047, #1052 .  Can you review those issues and see if that helps?

-Kevin

SergeyKrutyko, 2019-02-03:
> There was a long discussion on something similar with @trevjonez - see #916, #1044, #1047, #1052 . Can you review those issues and see if that helps?
> 
I read a lot of topics here. Unfortunately, I have a different project. This is not a port MMU2 - it is a fully Multi MCU project managed by a Klipper. I want to control the whole process with the Klipper tools. Therefore, I need a little help of your coding, or rather, how to properly perform certain tasks in the plugin correctly.

trevjonez, 2019-02-03:
I am indeed working to make klipper directly control the MMU2 it just happens that I am basing the work on the mmu2 PCB rather than the RAMPS (any mcu multiple or single would work). This is because many people have already purchased many instances of the MMU2 control board in the form of the prusa kits. Secondly, the 8bit MCU's are already loaded enough with managing their own steppers let alone being a relay for controls to an additional 3 steppers and filament sensor.

I do believe the approach you are taking with defining many extruders can work, but I believe it is the wrong abstraction for modeling what the MMU2 style hardware is.

if you give a look at [this branch on my fork of klipper](https://github.com/trevjonez/klipper/tree/tjones/mmu2_homing) I have homing working with the tmc2130's but could easily be adjusted to use any endstop so long as your hardware configuration works. It is important to note that MMU2 is specific to the mmu2 hardware using the mm2 control board as it does some hot patching of the host and reconfig of the IO in order to make the wonky GPIO config via shift registers work.

ips_filament_selector.py will be the most interesting to you for now.

I have not been able to dedicate much time to my MMU2 work due to other obligations, but I intend to have it working in Q1 2019 to some reasonable degree of success. Although it is important to keep in mind that there is a few other concerns with the MMU2 specific GPIO implementation that will need to be ironed out before I think a merge back to this repo can be feasible. (I have an idea in mind that I think will satisfy everyones goals without putting undue maintenance burden on the AVR line of mcu firmware as the current PR does.)

SergeyKrutyko, 2019-02-03:
trevjonez, You are well done! I like your work. My goal is to make a remix of mmu2. I want to use a good mechanic, but make the logic of the work in my own way. I do not think I consider the bad logic of the original device. I have an interest to implement my own algorithm. 

I want to make an affordable product, on a common element base

This is my hobby and a lot of people are waiting for my results.

drphil3d, 2019-02-04:
That's awesome, I have access to a ton of linear rods and lead screws if we wanted to build kits

SergeyKrutyko, 2019-02-05:
Trevjonez, thanks - its stupy error - replace tabs to speces and compile ok.... but not work (((((((((

SergeyKrutyko, 2019-02-05:
Trevjonez, I dont know - How to make a homing axis ... the source code is all dug up - everywhere complex dependencies...... where to take an example? This is simply procedure - but i dont know as :(

trevjonez, 2019-02-06:
yea that is pretty much what I ran into as well. I was able to get homing working and custom gcode hookups for my stuff to work. but I believe the correct solution is to make the axi config of gcode, toolhead, homing, composable. I started working through decoupling things and formalizing a contract of what an axi is and does, which also makes more sense for N axi machines or anything else for that matter. I think I will start fresh now that I have a better understanding of what all the various position values represent.  main pain point I see is that much of the klipper host code was written with the concept of XYZE being core and nothing else being relevant. so we can either workaround it being that way or rip up the floor and add some plumbing.

I fear it will be a big change list though so it may well need to take place over time. I hope to start on it tonight again. finally getting things caught up at home and work.

SergeyKrutyko, 2019-02-06:
Trevjonez, I have carefully studied your code, and noticed that you have seriously worked on the project.

I also thought that probably the best solution would be to change the kinematics XYZE -> XYZESIL

S - selector
F - iddler
L - loader

SergeyKrutyko, 2019-02-06:
I still have a question for you - if I perform a cycle for a long time, the MCU does not change the state of the buttons and does not trigger events on them. I understand me need to make a call to some kind of procedure, if a long cycle starts?

SergeyKrutyko, 2019-02-06:
```
class MMUSensor:
    def __init__(self, config, sensor_name):
            self.printer = config.get_printer()
            self.sensor_name = sensor_name
            self.pin = config.get('%s_pin' % (sensor_name))
            self.buttons = self.printer.try_load_module(config, "buttons")
            self.buttons.register_buttons([self.pin], self.sensor_change)
            self.status = 0
        

    def sensor_change(self, event_time, state):
        logging.info("SensorUpdate:%s: %i -> %i" % (self.sensor_name, self.status, state))
        self.status = state
        self.printer.send_event("%s:sensor_change" % (self.sensor_name), self, event_time, state)

    def get_status(self):
        return self.status
```

SergeyKrutyko, 2019-02-06:
```
#sensors setup
        self.sensor_s = MMUSensor(config, 'sensor_s')
        self.stop_s = self.sensor_s.get_status()
```

SergeyKrutyko, 2019-02-06:
and long cycle dont get endstop status:

```
def cmd_MMRU_HOME(self, params):
            force_move = self.printer.lookup_object("force_move")
            force_move.force_enable(self.s1)
            self.stop_s = self.sensor_s.get_status()
            while self.stop_s<1:
                logging.info("move move")
                self.get_toolhead().dwell(1)
                self.stop_s = self.sensor_s.get_status()
                #force_move.manual_move(self.s1, dist=-0.1, speed=1.0)
            logging.info("move END")
            self.gcode.respond_info("Home Selector Complete") 
```

trevjonez, 2019-02-06:
are you seeing the "SensorUpdate" in the logs? you should be able to do toggle that by hand and see it assuming it is a mechanical switch. if you look in the init it is setup as just a button which will call through to sensor_change.

SergeyKrutyko, 2019-02-06:
> are you seeing the "SensorUpdate" in the logs?
Yes, but in process i dont react on this button.... ((((

KevinOConnor, 2019-02-09:
I've put together some code on a new work-manual-stepper-20190208 branch.

cd ~/klipper ; git fetch ; git checkout origin/work-manual-stepper-20190208  ; sudo service klipper restart

This adds a new [manual_stepper] config section and a new MANUAL_STEPPER g-code command.  See config/example-extras.cfg and docs/G-Codes.md for the details.  With this support it should be possible to manually declare a stepper (that is separate from the normal robot kinematics) and manually control it via the MANUAL_STEPPER command.

-Kevin

SergeyKrutyko, 2019-02-09:
THANKS KevinOConnor!
You are Perfect!
that's what i need

SergeyKrutyko, 2019-02-11:
KevinOConnor, big thanks for manual stepper - i make own class bassed on your in my mmru plugin...

But now I have a new task:

There is a stepper, which has a enstop sensor. I need two modes of operation move to home:

1. First - until the sensor closes.
2. Second - until the sensor opens

Simply put - I need to change the sign of the signal inversion at an Enstop.

SergeyKrutyko, 2019-02-11:
I also often get an error at homming process: "Timeout during endstop homing" may be i must change parameters of timeout ?

KevinOConnor, 2019-02-12:
If you want to move until the pin is no longer triggered, then add a `triggered=False` parameter to the mcu_endstop.home_start() call.

You should only get a timeout error if the specified move completes without the endstop triggering.  Perhaps move a further distance in the homing move just to be sure?

-Kevin

SergeyKrutyko, 2019-02-12:
> You should only get a timeout error if the specified move completes without the endstop triggering. Perhaps move a further distance in the homing move just to be sure?

I did not understand the patterns, sometimes everything works fine, and sometimes it falls down with this error.

I first make a quick entry to the breakpoint, as soon as it is found, I take it back 5mm and again I search for the home more precisely, at a slow speed.

I move back to re-search for the end, I do not retract enough and the end-stop does not open. Did I understand you correctly?

KevinOConnor, 2019-02-12:
FYI, I've merged the manual_stepper code into the master branch.

It's hard to say what the problem is.  If your axis is 20mm long, I'd do homing with something like: `MANUAL_STEPPER SET_POSITION=30 MOVE=0 SPEED=10 STOP_ON_ENDSTOP=1` ; `MANUAL_STEPPER SET_POSITION=0 MOVE=5 SPEED=10` ; `MANUAL_STEPPER MOVE=-5 SPEED=1 STOP_ON_ENDSTOP=1` ; `MANUAL_STEPPER SET_POSITION=0`

-Kevin

KevinOConnor, 2019-02-25:
Any further updates on this?  If there are no further questions, I'm inclined to close this issue.

-Kevin

SergeyKrutyko, 2019-02-26:
@KevinOConnor ok - if i found new questions - i open new. Thanks!

closed 2019-03-04 by KevinOConnor