Klipper3d/klipper #1181: Adjust z height

forerunnert, 2019-02-01:
Hello, 
I want to adjust the z height of my delta printer 0.05mm down to dual in the first layer. 
What is the intended location to enter this offset? Preferably it should be persistent between prints.

megablue, 2019-02-01:
if i understood you correctly, 
you can use `SET_GCODE_OFFSET Z=-0.05` to dive closer to the bed, the offset will persistent between prints until you restart the controller board or the klipper service.  just add this on your startup gcode.

cmock, 2019-02-01:
alternatively, what I do is

<pre>
[homing_override]
gcode:
  G28
  SET_GCODE_OFFSET Z=.0.05
</pre>

forerunnert, 2019-02-01:
That's what I needed exactly. I found SET_GCODE_OFFSET in the docs but there is no mention of persistence over restart though so I thought I was looking at the wrong parameter.
I expected a configfile parameter.
Thanks!

Magoo0876, 2019-02-06:
What about the idea to tune the probes z offset? This would be persistent and no mod on the Gcode is needed.

KevinOConnor, 2019-02-08:
I'm closing this as I believe the original question was answered (use SET_GCODE_OFFSET).

-Kevin

KevinOConnor, 2019-02-14:
FYI, the new Bed_Level document describes how one can apply the SET_GCODE_OFFSET in a print start g-code macro: https://github.com/KevinOConnor/klipper/blob/master/docs/Bed_Level.md

-Kevin

closed 2019-02-18 by KevinOConnor