Klipper3d/klipper #1185: Temperature creeps up and print stops

stemaDev, 2019-02-02:
I have an ender 3 with an mks gen 1.4 board, since the one of the stock's drivers have failed. I set up everything, but when printing the calibration cube from petg @250°C but after 8-10 minutes the temperature starts to creep up and at one point it reaches 265, which I set as max_temp and it shuts down. I thought of thermistor and heater issues, but I didn't have issues with marlin and also if the temperature would go down, it would make sense that something's not working, but it's going up, so my best guess is the firmware. 
I would really like to use klipper, hopefully we can figure it out. Thanks :)

[image]

[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/2824197/klippy.log)

stemaDev, 2019-02-03:
Yes, several times, also tried with default and values from someone else with a similar setup

DanielJoyce, 2019-02-04:
I've seen this octoprint, which is why in the past I stopped using and went to other software.

stemaDev, 2019-02-04:
For me it might have been the board. Klipper was able to keep it close to the target for 5-10 minutes, but then it started to fluctuate and rise. I went back to marlin and though it was able to keep it around target, but was fluctuating near it with quite a few degrees. I changed thermistor, heater, but now I changed the board(was mks gen 1.4, now fysetc f6) and in marlin the temperature is super stable. I will try klipper if I have everything set up.

MarcinKnD, 2019-02-05:
My problem is solved.
the problem was in the failure of the extruder heater. it worked until the cooling fan of the printout turned on. The heater did not have the power to keep the temperature, it just wore out. Exchange for a new solution to the problem. Now the extruder is holding the temperature.

KevinOConnor, 2019-02-06:
@stemaDev - it looks like your extruder thermistor is very noisy - the raw temperatures are jumping around +-20 degrees.  That is causing the PID to not function correctly.  I'd double check your thermistor wiring to ensure it is not crimped or obstructed.

-Kevin

stemaDev, 2019-02-09:
Maybe it's the board, I'm not sure, but I changed it now, don't know about klipper yet, if I'll still have issues with the new one, I'll reopen the ticket. Thank you!

closed 2019-02-09 by stemaDev