Klipper3d/klipper #1194: Reported error: Section 'stepper_....' is not a valid config section on Multi MCU

SergeyKrutyko, 2019-02-04:
After Adding Config Section:

....
[mcu mmru]
serial: /dev/serial/by-path/platform-1c1c400.usb-usb-0:1:1.0-port0
pin_map: arduino

[stepper_sel]
step_pin: mmru:ar54
dir_pin: mmru:ar55
enable_pin: !mmru:ar38
step_distance: .000625

#endstop_pin: ^mmru:ar3
#position_endstop: 5

...........

Klipper dont restart and say error:

Reported error: Section 'stepper_sel' is not a valid config section

trevjonez, 2019-02-04:
the stepper suffix needs to match what is configured by the kinematics of your printer. typically _x, _y, _z. if nothing loads the config section it won't be considered valid as far as I understand

SergeyKrutyko, 2019-02-04:
In https://github.com/KevinOConnor/klipper/issues/700 Kevin write example and sat it must be work...

SergeyKrutyko, 2019-02-04:
i make dsimmru.py in extras folder:

import stepper

class DSIMMRU:
    def __init__(self, config):
        self.printer = config.get_printer()
        self.s1 = stepper.PrinterStepper(config.getsection("stepper_s"))

and make config:

[mcu mmru]
serial: /dev/serial/by-path/platform-1c1c400.usb-usb-0:1:1.0-port0
pin_map: arduino

[dsimmru]

[stepper stepper_s]
step_pin: mmru:ar54
dir_pin: mmru:ar55
enable_pin: !mmru:ar38
step_distance: .000625

now klipper say:

Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: Section 'dsimmru' is not a valid config section

trevjonez, 2019-02-04:
try just `[stepper_s]` because multi part config section names typically get loaded via `load_config_prefix` and make sure you defined the `load_config` function correctly for `dsimmru.py`

my guess is it failed to find stepper_s due to the prefixed config section you have and so dsimmru failed to init correctly.

SergeyKrutyko, 2019-02-04:
i forgot to add load_config in my py module

closed 2019-02-04 by SergeyKrutyko

kakou-fr, 2019-06-25:
> i forgot to add load_config in my py module

Hi SergeKrutyko. Some progress on the MMU support ?