Klipper3d/klipper #1195: Internal error on command:"G28"

ArdentSquirrel, 2019-02-04:
Running into an error when issuing "G28 XY".  Both X and Y axis home correctly, and then klippy shuts down with "Internal error on command:"G28".

I am attempting to use two mcu's, an ATMEGA2560 / RAMPS for all functions except Z, and then an Arduino Uno (328p) with the CNC shield for the Z motors.

This is a Piper2 CoreXY printer, with 4 separate Z motors.

klippy.log uploaded as requested.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2829390/klippy.log)

ArdentSquirrel, 2019-02-04:
I changed out the Arduino Uno with an Arduino AtMega2560, and the errors went away.  New klippy.log attached.
[klippy.log](https://github.com/KevinOConnor/klipper/files/2829850/klippy.log)

JohnEdwa, 2019-02-07:
@KevinOConnor Speaking of homing changes, would it be possible for you to also implement Safe Homing and Quick Homing from Marlin - Safe Homing lifts the Z up first before moving X or Y if the Z height is unknown or below a set limit (5mm by default) to clear binder clips, and Quick Homing moves both X and Y at the same time until one of them hits the endstop.
I know the safe homing can be kinda done with the override macro, but it can't be set to do the z-lift only when unhomed or below 5mm, and using the override breaks homing X and Y separately.

<sup><sub>Also, nice pun.</sub></sup>

KevinOConnor, 2019-02-08:
@JohnEdwa - No, I'm booked for the next year+ on other topics.  In general, if a new feature is desired it will require a developer to implement and submit as a github pull request.

That said, safe homing is possible via the homing_override config directive.  It is possible to limit the override so that it does not apply when homing xy - see config/example-extras.cfg for the details.

-Kevin

JohnEdwa, 2019-02-09:
>That said, safe homing is possible via the homing_override config directive. It is possible to limit the override so that it does not apply when homing xy - see config/example-extras.cfg for the details.

But you can't combine them, right?
Because with a moving bed, X or Y homing should always start with `if (Z < 5) Z = 5` (safe homing) because not doing so means if Y or X is already homed moving the other means you'll crash into the binder clips. 

The only solution I've come up with is
```
[homing_override]
set_position_z: 5
gcode:
    G1 Z10 F600 ; Move Z up
    G28 X0 Y0 Z0
```
and even if I could separate the axis, as it can't actually check the Z height but only lift it blindly, it wouldn't work anyway.

But as far as missing features or improvements go, this is a mild annoyance at best. Adding it to a 2020 maybe-todo list sounds about as urgent as it deserves to be :)

KevinOConnor, 2019-07-10:
FYI, this should now be fixed (PR #1766).

-Kevin

closed 2019-07-18 by KevinOConnor