Klipper3d/klipper #1197: Klipper shutdown: Error 'MCU 'mcu' shutdown: ADC out of range....'

klipartstudio, 2019-02-05:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2831413/klippy.log)

Hi everyone, 

I just updated to the latest Klipper from the repo. I seem to be getting an odd issue with Klipper shutting down during extruder heating.

I keep getting a error:
Recv: // MCU 'mcu' shutdown: ADC out of range
Recv: // This generally occurs when a heater temperature exceeds
Recv: // its configured min_temp or max_temp.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: !! MCU 'mcu' shutdown: ADC out of range

After this error I do a 'FIRMWARE_RESTART' which makes it work for about 5min and then shuts down again. Not sure if the temperature is fluctuating to much and causing it or what the deal is. If anyone has any insight on how to resolve this I would appreciate it. 

I also tested my extruder thermistor with my voltage meter temperature reader and it is achieving the right temps. For instance, when I set it to 215 it reaches that temperature sometimes.

Printer:
Cr10s, Micro Swiss All metal hot end

klipartstudio, 2019-02-06:
HI Team, 

Still have not figured out the issue. I am going to reinstall Merlin and see if it works. At least I know it. has nothing to do with the hardware and something to do either with octoprint or klipper.

[klippy.log](https://github.com/KevinOConnor/klipper/files/2834534/klippy.log)

trevjonez, 2019-02-06:
something is wrong with your extruder thermistor config/hookup. `extruder: target=215 temp=-46.9` min is probably set at 0 and I would guess that negative value is a floating input pin.

klipartstudio, 2019-02-06:
Hi trevjonez,

I tested my connection from cr10s to the clip and I was getting 5v; then I tested the thermistor resistance and it did not return 100k which I expected it was returning 1000-1400. So I changed the thermistor and everything worked. I honesty don't understand what went wrong cause that was a new working thermistor as well, if you see from my logs. This is why I thought it had something to do with Klipper. Either-way I am glad it is all working and klipper had nothing to do with it. I appreciate your response it definitely lead me to the right direction. Thank you!

closed 2019-02-06 by klipartstudio

xvart, 2020-12-02:
I cheated with this as I was testing and set my min-temp to -100 which allowed klipper to start and stay up

Sureiya, 2020-12-09:
Pretty sure this is a sign of a bad/faulty thermistor.