Klipper3d/klipper #1198: MKS LCD2004_ML support

bdn76, 2019-02-05:
What about MKS LCD2004_ML (two line, text based) support with MKS Gen_L V1.0

[image]
[image]

?

KevinOConnor, 2019-02-06:
That appears to be a standard "RepRapDiscount 2004 Smart Controller" display - see the section on that in the config/generic-ramps.cfg file.

-Kevin

bdn76, 2019-02-06:
Ok, i'm try this later. In which case should the display print meaningful information?
1. Immediatly after start module.
2. Only after connect klippy to module and stop displaying after disconnect.
3. Only after start printing and stop displaying after printing end.

In other words, what is the degree of autonomy for displaying information?

bdn76, 2019-02-06:
Sorry, but not work:
[image]

KevinOConnor, 2019-02-07:
In order for us to assist, you'll need to attach the Klipper log file - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

bdn76, 2019-02-08:
Sorry, no problem.
[log.zip](https://github.com/KevinOConnor/klipper/files/2844353/log.zip)

KevinOConnor, 2019-02-08:
I don't know why the display did not work.  Generally (after a full power reset) the display wont show anything until it is initialized.  So, it is odd that it is showing some data.  I'd double check the wiring.  Otherwise I don't have any additional advice.

-Kevin

bdn76, 2019-02-09:
I'm check wires - all wires are correct.

1. Can i send string to display via cli before starting print?
2. Can you modify MCU firmware for print something to display immediatly after firmware start (idealy is a version and time elapsed after start with 1 secod update). This is simplify wire problem issue.

bensen-666, 2019-02-12:
Yes, Same Problem with my 2004 display.

KevinOConnor, 2019-03-11:
Any further luck with this?

-Kevin

bdn76, 2019-03-11:
No. I'm disconnect and power off disaplay.

KevinOConnor, 2019-04-07:
Okay, I'm going to close this then.  It is not known why it is not working and it doesn't look like there are any leads to getting it to work.

-Kevin

closed 2019-04-07 by KevinOConnor

humbuckr, 2019-07-05:
I recently had my stock Ender 3 main board lose its ability to control the bed temperature.  Googling lead me to believe i needed a new main board.  I got the MKS Gen 1.4 and the TMC2208 Stepper drivers.  

All I see on the STOCK LCD is the **backlight** after repeated firmware updates.

Things I've tried - 

- I've flipped the ribbon cable around
(flipped around on the slot 3 of the LCD is the only location where it lights up the LCD
- commented out // #define CR10_STOCKDISPLAY in configuration.h AND
- //#define REPRAP_DISCOUNT_SMART_CONTROLLER (tried both on and off)
- Swapped out the MKS Gen 1.4 with the stock creality board and the LCD displays the Ender menus as normal

Under a deadline to prints so I'm frustrated.    Would appreciate any tips.  thanks