Klipper3d/klipper #1200: MKS GEN L an 7920 ender 3 stock display

bensen-666, 2019-02-05:
The logfile will not.help, because there is no error. I want to use stock 7920 ender 3 display. But it didnt work. The pinout from the mks ramps 1.3 board seems wrong. With a full graphics display it works. 

When i comment the original 7920 i dont see anything. When i put the cable 180° turned the display background lights. But is see nothin. The display is clear from signs..

Magoo0876, 2019-02-05:
I'm using the stock display from the ender 3 type 12864. The connector at the display has to go to exp3, on the GEN L to EXP1 and the stock cable must be turned round. My settings are: 

[display]
lcd_type: st7920
cs_pin: ar27
sclk_pin: ar25
sid_pin: ar29
encoder_pins: ^ar23, ^ar17
click_pin: ^!ar35

bensen-666, 2019-02-06:
Oh shit! I'm very dumb... I mean ENDER-4, not Ender 3 thats another display in Ender-4... Its smaller....in height.

Magoo0876, 2019-02-06:
It is difficult to support you if you don't know what you are using. Maybe the pictures below can help. The 12864 LCD uses the st7920 as controller. I still recommend to check with a Terminal if the Printer Is ready. If there is a small other thing you will never see more than the backlight. 

The 2004 has 20 columns 4 rows. 
[image]

ST7920 128x64 dots. 
[image]

bensen-666, 2019-02-07:
[image]

So i took a picture. Left display works, when i comment it in printer.cfg original display right is from ender 4 with 4 rows. I cant bring it to work

KevinOConnor, 2019-02-07:
In order for us to assist, you'll need to attach the klipper log file from - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

bensen-666, 2019-02-08:
I hope thats what you need

[klippy.log](https://github.com/KevinOConnor/klipper/files/2846420/klippy.log)

bensen-666, 2019-02-08:
With this preferences i see following on my ender-4 2004 display....
[image]

Magoo0876, 2019-02-08:
I'm sorry but I do not have such a display to support you better. Fortunately there is already an issue to face this problem.  #1198 
Maybe you can use the other display in the meanwhile if you really need one.

bensen-666, 2019-02-09:
I thank you for your time Magoo0876..... I read other issues to get an information...... But Kevin don't seem to know why it doesn't work... I checked the link..... But there is no solution

bensen-666, 2019-02-09:
[image]

Now i have some funny things on my display 😀

bensen-666, 2019-02-10:
[klippy.log](https://github.com/KevinOConnor/klipper/files/2849116/klippy.log)

Is this an Failure of translation of the ISO and UTF standarts??

KevinOConnor, 2019-02-18:
As with issue #1198, I don't know what would cause the above.  Best suggestion would be to verify wiring and pin definitions.

-Kevin

KevinOConnor, 2019-03-11:
I'm going to close this in favour of the existing issue #1198.

-Kevin

closed 2019-03-11 by KevinOConnor

provistapp, 2019-11-10:
Kevin,
I have the same problem, never solved.
Please advise model LCD to buy that will work with MKS GEN L V1.0 and config in Marlin,
Thanks