Klipper3d/klipper #1201: Negativ Mesh Value

schnello, 2019-02-06:
Hello.

Iam sorry to open an issue regarding my question but i study the doc and i doenst found the answer.

I setup the [bed_mesh] function without an sensor (manual) and it works fine. Only one problem: The system doesnt accept any value below zero. So what is now the recommended way to "solve" the problem? Below the values from my bed with marlin MBL:

G29 S3 X1 Y1 Z-0.02000
G29 S3 X2 Y1 Z0.02000
G29 S3 X3 Y1 Z0.06000
G29 S3 X4 Y1 Z0.04000
G29 S3 X5 Y1 Z0.02000
G29 S3 X1 Y2 Z-0.02000
G29 S3 X2 Y2 Z0.06000
G29 S3 X3 Y2 Z0.06000
G29 S3 X4 Y2 Z0.06000
 G29 S3 X5 Y2 Z0.00000
G29 S3 X1 Y3 Z-0.02000
G29 S3 X2 Y3 Z0.02000
G29 S3 X3 Y3 Z0.06000
G29 S3 X4 Y3 Z0.06000
 G29 S3 X5 Y3 Z0.00000
G29 S3 X1 Y4 Z-0.02000
G29 S3 X2 Y4 Z0.02000
G29 S3 X3 Y4 Z0.04000
G29 S3 X4 Y4 Z0.04000
G29 S3 X5 Y4 Z-0.02000
G29 S3 X1 Y5 Z-0.04000
G29 S3 X2 Y5 Z0.02000
G29 S3 X3 Y5 Z0.04000
G29 S3 X4 Y5 Z0.04000
G29 S3 X5 Y5 Z-0.04000

Thanks a lot,
Christian

bshowket, 2019-02-06:
did you try with negative value in cfg file for Z axis, such as
 
`position_endstop: -2.0`

schnello, 2019-02-06:
Hello.

If i understand the position_endstop value correct it will change the zero position. If i change the zero position the basic leveling (the 4 corners) are failed. 

In the meanwile i was looking in the standard config (I was too fixated on the extra config) and i found the value "position_min:". I hope this can fix the problem.

schnello, 2019-02-06:
Looks the "**position_min**:" was the solution.

Thanks a lot.

closed 2019-02-06 by schnello