Klipper3d/klipper #1205: MKS GEN L Ender 3 TMC2208 UART strage dump

Magoo0876, 2019-02-06:
I've read almost everything I could find about the TMC 2208, but still there are some questions. In principal the printer is moving, but moves differently compared to others @ youtube. 
The settings are: current 0.725A and 0.9 for the extruder, 100mm/s modeswitch.
- Is it ok to set MS1 and MS2 to set 16 step mode also with UART connection. With MS3 closed there is no Uart connection. Without any jumpers I had strange noises like an opened water pipe.
- What is the right VREF for UART and does it interfere with the software current settings?
- Please look at my TMC_Dump, the coil currents differ a lot and the OFS is high. Is there something wrong ?

Thanks a lot in advance.

Send: DUMP_TMC STEPPER=stepper_x
Recv: // GCONF:      000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
Recv: // GSTAT:      00000001 reset=1
Recv: // IFCNT:      0000004e IFCNT=78
Recv: // OTP_READ:   0000000d OTP_FCLKTRIM=13
Recv: // IOIN@TMC220x: 2000034c MS1=1 MS2=1 PDN_UART=1 SEL_A=1(TMC220x) DIR=1 VERSION=0x20
Recv: // FACTORY_CONF: 0000000d FCLKTRIM=13
Recv: // TSTEP:      000fffff TSTEP=1048575
Recv: // MSCNT:      00000238 MSCNT=568
Recv: // MSCURACT:   011701ac **CUR_A=-84 CUR_B=-233**
Recv: // CHOPCONF:   14030053 toff=3 hstrt=5 TBL=2 vsense=1 MRES=4(16usteps) intpol=1
Recv: // DRV_STATUS: c0090000 CS_ACTUAL=9 stealth=1 stst=1
Recv: // PWMCONF:    c80d1b65 PWM_OFS=101 PWM_GRAD=27 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
Recv: // PWM_SCALE:  0000001f PWM_SCALE_SUM=31
Recv: // PWM_AUTO:   001e0064 **PWM_OFS_AUTO=100 PWM_GRAD_AUTO=30**
Recv: ok
[...]
Send: DUMP_TMC STEPPER=stepper_y
Recv: // GCONF:      000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
Recv: // GSTAT:      00000001 reset=1
Recv: // IFCNT:      0000004e IFCNT=78
Recv: // OTP_READ:   0000000f OTP_FCLKTRIM=15
Recv: // IOIN@TMC220x: 2000034c MS1=1 MS2=1 PDN_UART=1 SEL_A=1(TMC220x) DIR=1 VERSION=0x20
Recv: // FACTORY_CONF: 0000000f FCLKTRIM=15
Recv: // TSTEP:      000fffff TSTEP=1048575
Recv: // MSCNT:      000000e8 MSCNT=232
Recv: // MSCURACT:   002300f4 **CUR_A=244 CUR_B=35**
Recv: // CHOPCONF:   14030053 toff=3 hstrt=5 TBL=2 vsense=1 MRES=4(16usteps) intpol=1
Recv: // DRV_STATUS: c0090000 CS_ACTUAL=9 stealth=1 stst=1
Recv: // PWMCONF:    c10d1263 PWM_OFS=99 PWM_GRAD=18 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=1 PWM_LIM=12
Recv: // PWM_SCALE:  0001001f PWM_SCALE_SUM=31 PWM_SCALE_AUTO=1
Recv: // PWM_AUTO:   00140060 **PWM_OFS_AUTO=96 PWM_GRAD_AUTO=20**
Recv: ok
Send: DUMP_TMC STEPPER=stepper_z
Recv: // GCONF:      000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
Recv: // GSTAT:      00000001 reset=1
Recv: // IFCNT:      0000004e IFCNT=78
Recv: // OTP_READ:   0000000e OTP_FCLKTRIM=14
Recv: // IOIN@TMC220x: 2000014c MS1=1 MS2=1 PDN_UART=1 SEL_A=1(TMC220x) VERSION=0x20
Recv: // FACTORY_CONF: 0000000e FCLKTRIM=14
Recv: // TSTEP:      000fffff TSTEP=1048575
Recv: // MSCNT:      000001b8 MSCNT=440
Recv: // MSCURACT:   011f0068 **CUR_A=104 CUR_B=-225**
Recv: // CHOPCONF:   14030053 toff=3 hstrt=5 TBL=2 vsense=1 MRES=4(16usteps) intpol=1
Recv: // DRV_STATUS: c0090000 CS_ACTUAL=9 stealth=1 stst=1
Recv: // PWMCONF:    c80d1b68 PWM_OFS=104 PWM_GRAD=27 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
Recv: // PWM_SCALE:  00030022 PWM_SCALE_SUM=34 PWM_SCALE_AUTO=3
Recv: // PWM_AUTO:   00200064 **PWM_OFS_AUTO=100 PWM_GRAD_AUTO=32**

Send: DUMP_TMC STEPPER=extruder
Recv: // GCONF:      000001c0 pdn_disable=1 mstep_reg_select=1 multistep_filt=1
Recv: // GSTAT:      00000001 reset=1
Recv: // IFCNT:      00000054 IFCNT=84
Recv: // OTP_READ:   0000000c OTP_FCLKTRIM=12
Recv: // IOIN@TMC220x: 2000014d ENN=1 MS1=1 MS2=1 PDN_UART=1 SEL_A=1(TMC220x) VERSION=0x20
Recv: // FACTORY_CONF: 0000000c FCLKTRIM=12
Recv: // TSTEP:      000fffff TSTEP=1048575
Recv: // MSCNT:      00000008 MSCNT=8
Recv: // MSCURACT:   00f7000c **CUR_A=12 CUR_B=247**
Recv: // CHOPCONF:   14010053 toff=3 hstrt=5 TBL=2 MRES=4(16usteps) intpol=1
Recv: // DRV_STATUS: c0050000 CS_ACTUAL=5 stealth=1 stst=1
Recv: // PWMCONF:    c80d0e24 PWM_OFS=36 PWM_GRAD=14 pwm_freq=1 pwm_autoscale=1 pwm_autograd=1 PWM_REG=8 PWM_LIM=12
Recv: // PWM_SCALE:  00000006 PWM_SCALE_SUM=6
Recv: // PWM_AUTO:   000e0024 **PWM_OFS_AUTO=36 PWM_GRAD_AUTO=14**

KevinOConnor, 2019-02-07:
It's not clear to me what you're reporting.  It is normal for the fields you highlighted to change as the motor moves to different positions.  Beyond that, you will need to attach the klipper log file - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

Magoo0876, 2019-02-08:
I've already found out that push and pull is an normal operation what can cause opposite currents on the coils. According to Trinamic's docs there shouldn't be an influence from the ENN, DIAG, INDEX, MS1, MS2 Jumpers and Vref Value to the software settings. So maybe it is a good idea to set vref to an idle current value to prevent too high current if the host is not connected. In my case the jumper MS3 disables the Uart and enables water pipe sound. So its strange to have MS3 on the 2208 step stick. Maybe this will help someone with the same behavior ore questions.

closed 2019-02-08 by Magoo0876