Klipper3d/klipper #1207: CNC Router?

keeper317, 2019-02-07:
Would it be possible to configure klipper to run a cnc router instead of a 3d printer? I am looking to build a router and have been using klipper on my printer for a while and would like to continue to use it on a router.

gdachs, 2019-02-07:
Why not? Just don't configure extruder and heaters.

trevjonez, 2019-02-07:
It would need some code work as far as I know the main klipper toolhead implementation assumes there will always be 1 or more extruders configured.

that said you could probably give it some dummy config and move forward as is. but for best results new implementation would be required.

KevinOConnor, 2019-02-07:
I know a few people that have done that.  I'm not personally familiar with cnc machines, so can't say what would work well or wouldn't work well.  For laser cutters, be aware of the discussion in issue #133.

FYI, Klipper does not require an extruder to be configured.  If you don't have an extruder, then just don't define that section.

-Kevin

keeper317, 2019-02-08:
If I don't setup an extruder how would I setup the spindle motor control in the config?

Hywelmartin, 2019-02-08:
@keeper317  If the control for the spindle is a PWM signal use the pin assignment

https://github.com/KevinOConnor/klipper/blob/6c10e4bcef874991a7a6ac741eb4a88b41d9d150/config/example-extras.cfg#L601

and then make gCode macros to support M4 and M5
https://github.com/KevinOConnor/klipper/blob/6c10e4bcef874991a7a6ac741eb4a88b41d9d150/config/example-extras.cfg#L1142

Papsnoek, 2019-02-08:
Just FYI, I am running Klipper on my MPCNC without any issues. I Originally did this for the S-Curve Acceleration, but didn't see much of a change in behavior, so reverted back to standard config.

Everything works perfectly, except for the "Not moving without Homing" issue. I get around this by using a custom Macro for my Homing Config, to just force the machine to assign "Home" at where I position the Axis. 
Then I can jog the machine around from Octoprint, and issue an G92 X0 Y0 Z0 Command when Im happy with my "New Home Position"

KevinOConnor, 2019-02-18:
I'm going to close this issue as I think the original question has been answered (yes, it's possible, but some work may be needed depending on the g-code flavour being used).

-Kevin

closed 2019-02-18 by KevinOConnor

geoff-coppertop, 2019-03-03:
> Just FYI, I am running Klipper on my MPCNC without any issues. I Originally did this for the S-Curve Acceleration, but didn't see much of a change in behavior, so reverted back to standard config. [...]

Would you be willing to share your config?

Papsnoek, 2019-03-27:
Hi @geoff-coppertop 

Sorry for the very late reply, I haven't been using my CNC lately. 
My config is setup for 2 steppers per X and Y Axis, and I have added a custom Homing_Override so I can Home, and then Move my Gantry where I like. I hae  also set the bed size to be massive in positive and negative Directions, so I can move it where I like without limits.

Basically, I know what im doing, so I config the Printer to allow ME to do what I would like :)

[printer.zip](https://github.com/KevinOConnor/klipper/files/3014748/printer.zip)

greyhound3, 2020-04-14:
> Basically, I know what im doing, so I config the Printer to allow ME to do what I would like :)
Could you also post your macros, where G4 and G5 is defined? Thanks!

Papsnoek, 2020-04-14:
Hi @greyhound3 ,

My CNC raspberry Pi is currently re-purposed :) So i cannot access any files right now.

However, I didnt use any G4 or G5 Macros. I just created two Buttons in Octoprint or CNCJS.
1. Reset XY as G92 X0 Y0
2. Reset Z as G92 Z0

So I would "Home" my machine anywhere, jog it to my start point and "Reset XY", then I would position my Bit over the work piece and lower it until it touched, and then run my "Reset Z". This in theory would give me a my start XYZ for the work piece and I would cut away.

greyhound3, 2020-04-15:
Thanks a lot for your answer! I have to wait until the pwm to 10V power regulating board arrives and will report back and hopefully klipper is able to fulfill my cnc needs/dreams.
Anyhow: Thanks everybody and KevinOConnor especially a lot for klipper !!!

th3fallen, 2021-07-20:
@Papsnoek sorry for the really late ping but were you able to get cncjs to connect to klipper? if so could you share how?

Papsnoek, 2021-07-21:
Hi @th3fallen, no worries. 
I don't know if I can classify it as "working" :) I can get CNCJS to Connect, but there was so many unrecognized commands, that I didnt really continue using CNCJS. I ended up using Octoprint. The way I see it, CNCJS is much better with CNC and Octoprint is better with 3D Printers, and Klipper doesnt support much of the CNC CGode commands.

However, if you want to play a bit (at your own risk :) ), to get it working, you need to edit the file "~/.cncrc", then add to the file 
```
"ports": [
{
"comName": "/tmp/printer",
"manufacturer": "Klipper"
}
]
```

This will allow you to then pick the virtual serial port to connect to Klipper. Once you have saved the file and reloaded CNCJS, you can select the Serial Port and Connect as "Marlin".

Immediately, you will see you cannot move the machine until you home it, however, the default Home commands in CNCJS are unsupported by Klipper, so you need to add a Command for G28 to Home. so you basically end up writing alot of Custom Commands/Buttons to get things going :)

Let me know if you managed to get this working, and if you do decide to use CNCJS, let me know if you find it better?

vladbabii, 2021-07-28:
Have you seen my try at running klipper as cnc ? I'm working on a new version (rebuilding my cnc) but until then this has been proven to work - https://github.com/vladbabii/klippy-cnc

vladbabii, 2021-07-28:
I'm open to revisit anything on this implementation for version 2, so if you have any ideas feel free to let me know.