Klipper3d/klipper #1213: Ender 3 - Firmware not booting anymore

savage7, 2019-02-08:
I've an Ender 3 with stock board, I flashed it successfully and had Klipper running already.
Then I went back to Marlin, installed BLTouch Clone and used it.

Went back to Klipper to get it working with BLTouch, while probing the nozzle crashed into the bed and I powered off the printer.

From this moment the firmware didnt boot. I flashed Marlin again:  Marlin is working.
Flashed Klipper again: Klipper is not booting
Flashed the bootloader again and then klipper: not booting
Deleted klipper from my Pi, did a fresh git pull and build: not booting
double checked every config: Didn't find anything suspicious 

With not booting I mean:
- Nothing shown on the display of the printer (before I had a menu)
- No connection possible via octoprint

When I flash it finishes successfully, but never boots. I also checked many times if I flash to the correct /dev/ttyUSB0 device, also used the real name. But flashing works all the time. 

Very strange behaviour. I'm not sure if this is a klipper issue, but I dont have any idea left what to try :( 

Has anyone an idea, what the problem could be? 

Thank you!

cosminr86, 2019-02-08:
We need klippy.log

savage7, 2019-02-08:
Oh my what stupid mistake. Just tried to copy the klippy.log and found the error. Simple config error which was shown in the log. 
Sorry for bothering.

closed 2019-02-08 by savage7