Klipper3d/klipper #1215: hotend temperature become increasingly unstable despite auto pid tuned and low fan speed

megablue, 2019-02-08:
[image]

i am using mks gen l as my controller board. as title, i have my hotend pid auto tuned at 215C with 50% fan speed for the part cooling fans.  i've installed a home made silicone sleeve for the heater block, i've the part cooling fans turned on at layer 2, only at 15% fan speed, which barely pushes any air but it is already enough to cause crazy fluctuations and it get worse as it goes on, at worse the temp will go up and down for +5/-5C

 it might be worth mentioning that i've tried setting up various conditions prior auto pid tuning, such as, sleeve on/off, various fan speed, various z height. however, nothing changes, the thermal instability kicks in as soon as there are slightest amount of breeze over the heater block. 

is there anything i can do to improve the temperature stability for my hotend?

KevinOConnor, 2019-02-08:
It definitely looks like poor pid parameters.  I'm not sure why PID_CALIBRATE isn't working well for you.  Could you rerun PID_CALIBRATE and then post the log from that session?

-Kevin

megablue, 2019-02-09:
here is the log for PID_CALIBRATE
[Uploading klippy.zip…]()

KevinOConnor, 2019-02-10:
I don't know why that would occur.  It doesn't appear to be a defect in Klipper - my best guess is that the heater just isn't getting good pid parameters.  If you're having an issue with the heater stability with the part cooling fan at 15% then you could try performing the pid tune with the fan at that speed.

-Kevin

megablue, 2019-02-12:
i've tried, pid tuned at 15% fan speed specifically and even stick to the pid tuned temp. am still getting the same issue.

KevinOConnor, 2019-02-13:
I'm not sure why that would be.  Some users have reported success reducing the max_power (eg, max_power=0.7).  It may require manual tuning of the pid though.

-Kevin

megablue, 2019-02-14:
alright, thanks anyway. i will experiment with a different board to verify if the board is the culprit.

KevinOConnor, 2019-02-25:
Were you able to find a resolution to this?

-Kevin

megablue, 2019-02-25:
i've tried max_power at 0.7 it works much better, but only at the specific temperature that i pid tuned.  so...no not really.

vopt, 2019-03-08:
It could be a component fault in the Gen L board. Some have reported fixing this problem by adding or replacing low ESR capacitors around the switching regulator. I’m still looking into it because I have the same issue after swapping a Gen L into my Ender 3.

KevinOConnor, 2019-04-01:
Well, it doesn't look like anyone has a solution, so I'm going to close this.  We can reopen if there is further ideas.

-Kevin

closed 2019-04-01 by KevinOConnor

bjones14, 2020-01-03:
I think this is related and/or could be resolved by PID gain scheduling.  I too am having issues tuning my PID when the fan is on, and keep getting an error from the thermal protection inherit to Klipper.

See this issue - https://github.com/KevinOConnor/klipper/issues/1167

Ideally there would be a mechanism to change the PID parameters based on scheduling variables (such as layer fan at 15% speed or temperature).  There just simply isn't a single PID that is going to provide accurate control for all these different conditions.