Klipper3d/klipper #1218: MCU lost communication and heater still active

Doni2308, 2019-02-08:
Hello, I do not know what to do next. My printer loses the connection to the MCU shortly after starting a print job. I have already tried 4 different USB cables, 3 different power supplies and yet the error occurs again and again. What worries me is that both the heating bed and the extruder remain active. I hope someone can find the error from the log file.
[klippy.log](https://github.com/KevinOConnor/klipper/files/2847077/klippy.log)

KevinOConnor, 2019-02-11:
There's nothing in the log that provides any further details.  See the FAQ wrt lost communication errors: https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#i-keep-getting-random-lost-communication-with-mcu-errors

As for the heaters/extruders being on - have you run the checks described in https://github.com/KevinOConnor/klipper/blob/master/docs/Config_checks.md ?

-Kevin

Doni2308, 2019-02-11:
I guess I found the issue for the lost communication. I had a bad grounding in the supply cable to my power supply. After I fixed that no more fault occured. 
Regarding the active heaters. If i send a M112 command or if I disconnect the USB cable the heaters stop after some time. But if I remove the grounding of the power supply and the communcation get lost the heaters remain active. 
(Power Supply is a Phoenix Quint 20A)

DanielJoyce, 2019-02-11:
Sounds like there needs to be a timeout?

KevinOConnor, 2019-02-11:
That sounds like a hardware issue (with nothing the software can do to avoid it).  FWIW, I know the duet2 hardware adds physical components to the board to prevent grounding issues that could enable the FETs.

-Kevin

Doni2308, 2019-02-12:
As this seems to be a Hardware issue I am going to close this issue.

closed 2019-02-12 by Doni2308