Klipper3d/klipper #1219: Display glitches after updating Klipper

FeuerVogel666, 2019-02-09:
Yesterday I have updated Klipper, so the Part on the Raspberry and the part on the Arduino.
I have an Ender3 and a Ramps setup wich is soldered to 24Volt with a Converter.

After a restart evrything looks fine and as espected, but after a little time it shifts the Layers around and gets more and mor unusable.
That goes on, till there are only these big fat Rings, on the screen.

[image]
[image]
[image]

KevinOConnor, 2019-02-11:
In order for us to assist, you will need to attach the klipper log file - see: https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

closed 2019-02-18 by KevinOConnor

FeuerVogel666, 2019-02-19:
I would post the log file, but there isnt any. I´ve searched for klippy.log and found nothing, even there is no Folder named tmp.
Do I have to aktivate logging?

jakep82, 2019-02-19:
/tmp is in your root directory.

FeuerVogel666, 2019-02-19:
Thanks, I have found it.
I have figured out, that the problem with the display starts, when the printer is starting with the probing of the bed.
In Atachment my log File.
[klippy.log](https://github.com/KevinOConnor/klipper/files/2882379/klippy.log)

FeuerVogel666, 2021-01-27:
The Problem still exists. Yesterday i made a complete rebuild from the Printer (moved the electronics to a new Case and had a broken SD Card from the Raspberry Pi. So I made a clean reinstall.
And the Display still makes this Problem. Also I havent any Probe anymore, but a normal z-Endstopswitch.
Its a Stock Ender 3, except the MKS Gen L.
In Atatchment Below the new log.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5882685/klippy.log)