Klipper3d/klipper #1220: Feature request: detect lost steps on homing

cmock, 2019-02-09:
What I'd like is a variation of the homing command which can tell me if the printer has lost steps since the last homing (provided the motors have not been turned off since).

The use case would be to tune all kinds of parameters like accelerations, stepper currents, TMC driver parameters and whatnot, by homing, running a series of moves, and checking with this new command whether there were lost steps. 

At the moment, I may be able to visually detect massive step loss, but for anything on the scale of 1mm or less, the only way to see it would be a real print with the associated waste of filament.

KevinOConnor, 2019-02-10:
I'm not really sure what you are proposing.  If you'd like to know the number of steps the stepper motor has taken since the last home then that can be done via the GET_POSITION command.  In brief, home the printer, run GET_POSITION, note the "mcu:" positions for each stepper, run some series of movements, home the printer again, run GET_POSITION again, and ideally the "mcu:" positions should be unchanged.

Note that endstop switches themselves tend to trigger at slightly different positions, so a difference of a couple of microsteps is likely the result of endstop inaccuracies.  A stepper motor itself can only lose steps in increments of 4 full steps.  (So, if one is using 16 microsteps, then a lost step on the stepper would result in the "mcu:" step counter being off by a multiple of 64 microsteps.)

-Kevin

cmock, 2019-02-11:
If it works that way, that's fine, and I'll give it a try this evening.

I'll try to write up a short description of the process for the FAQ if it works out.

cmock, 2019-02-16:
Thanks, this works just fine on the CoreXY and the Delta.

There is one weirdness however: on the Delta, I get different "mcu" readings after the first G28 vs all later G28 commands, they differ by the value of my `GCODE_OFFSET`:

That is, I (re)start klipper, do a `G28` and a `GET_POSITION` and get

    // mcu: stepper_a:246 stepper_b:206 stepper_c:246

After another `G28`:

    // mcu: stepper_a:18 stepper_b:-21 stepper_c:18

After the third `G28`:

    // mcu: stepper_a:18 stepper_b:-22 stepper_c:18

And then it stays at those numbers. The differences between the values (246-18 etc) match the `GCODE_OFFSET` (in Z only) I set in my `[homing_override]` section. The stepper, kinematic, toolhead and gcode values behave accordingly, the gcode base and gcode homing values however do not change.

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/2871936/klippy.log.zip)

KevinOConnor, 2019-02-18:
That's really odd.  Note that the "mcu:" position doesn't report a valid result until after the first G28 completes.  I don't know why it would report a different value after the first G28 though.

-Kevin

cmock, 2019-02-18:
My impression is that it needs an absolute move or another `G28` after that first `G28` to apply the `GCODE_OFFSET`; relative moves (as in my `[homing_override]`) don't have that effect, I think I also tried some relative moves via the terminal.

KevinOConnor, 2019-02-18:
The "mcu:" position looks correct - what I think is causing your confusion is that the amount of relative movement changes when you issue the SET_GCODE_OFFSET the first time.  (The first time you run it, the gcode processor will add 2.85mm of Z to the next move, so the following relative move only actually moves 2.15mm down.)  So, either run SET_GCODE_OFFSET prior to G28 or run GET_POSITION prior to SET_GCODE_OFFSET.

-Kevin

KevinOConnor, 2019-03-04:
Thanks.  The FAQ update was committed in 82efed5e.

-Kevin

closed 2019-03-04 by KevinOConnor