Klipper3d/klipper #1221: Z Probe Homing On Delta

JarradFairborn, 2019-02-10:
I have started using a piezo z-probe and am getting great accuracy and repeatability. My main source of Z height inaccuracy has become my endstops. I was thinking of using the endstops for x,y homing only then using z probe to home Z height. It seems like the homing override as they are would not accomplish this.

Thanks
Jarrad

megablue, 2019-02-10:
if the repeatability of the endstops are really bad. even if you are able to get a proper z height with the zprobe, as soon as it starts moving away from the (assumed) center, the effector will not able to maintain its parallelism with the bed due the unknowable stage of the carriages.

Hywelmartin, 2019-02-10:
have you tried this [Stepper phase adjusted endstops](https://github.com/KevinOConnor/klipper/blob/cd6a18142908738d83590aa6e5af2d191055bfdf/config/example-extras.cfg#L400)

KevinOConnor, 2019-02-10:
Klipper doesn't currently support running DELTA_CALIBRATE at the start of each print.  If one were just concerned about bed height, then one could run a `PROBE` followed by a `G92 Z0`, but as mentioned above, that wouldn't account for differences between tower switches.  Also as mentioned above, you may wish to consider https://github.com/KevinOConnor/klipper/blob/master/docs/Endstop_Phase.md

-Kevin

JarradFairborn, 2019-02-11:
I will give Endstop_Phase a try. The problem I see with that is that I am not using a Raspberry PI. I will see if I can configure my system to reset the controller properly.

closed 2019-02-11 by JarradFairborn