Klipper3d/klipper #1223: BLtouch (genuine) Unknown command:"SET_SERVO"

bruce356, 2019-02-10:
Updated firmware 16/02/2019 
and get the following:-
Recv: // Unknown command:"SET_SERVO"
Is this no longer needed/valid with the new BLtouch config.
Thanks - bruce

[klippy.log](https://github.com/KevinOConnor/klipper/files/2848845/klippy.log)

KevinOConnor, 2019-02-10:
When using the bltouch module, one can use the BLTOUCH_DEBUG command to send commands to the bltouch.  See https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md

-Kevin

bruce356, 2019-02-11:
Thanks for the rep[y, there is nothing wrong with the BLtouch it functions fine.

With the original setting up of the BLtouch functionality using [servo bltouch] and [probe] one had to "SET_SERVO SERVO=bltouch ANGLE=xx". 

With the new configuration of the BLtouch I have now removed "SET_SERVO" from [homing_override].
Thanks again Regards - bruce

closed 2019-02-12 by bruce356