Klipper3d/klipper #1224: Display not work after update

semos2k, 2019-02-10:
I have updated to the latest version of klipper and my screen has stopped working.
I did tests with two boards mks sbase and skr in both I obtained the same result.

[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/2848884/klippy.log)

KevinOConnor, 2019-02-11:
Odd.  We haven't made any changes to the display code recently.  There was some recent optimizations made to the lpc176x code though.

There are a couple of things you could try to trouble shoot this.  First is to go to a prior version of the code, reflash the micro-controller, and see if it fixes the problem:
cd ~/klipper ; git checkout c171b28a0b ; make clean ; sudo service klipper stop ; make flash ; sudo service klipper start

The second thing you could do is to see if increasing ST7920_CMD_DELAY and/or ST7920_SYNC_DELAY at the top of ~/klipper/klippy/extras/display/st7920.py fixes the issue.  (Be sure to run `sudo service klipper restart` after any code change.)

-Kevin

semos2k, 2019-02-11:
@KevinOConnor when checkout the revision [c171b28] worked again. 

thx.

semos2k, 2019-03-03:
Fixed on last revision.

thx.

closed 2019-03-03 by semos2k