Klipper3d/klipper #1229: Need help with bltouch setup

CharlesStArnault, 2019-02-10:
While setting up bltouch in my config file, I ran in a problem. I get this error message in octoprint

Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: Unable to translate pin name: config_end_stop oid=14 pin=z_virtual_endstop pull_up=0 stepper_count=1

I've attached my config. Please help me!
[configBLTOUCH.txt](https://github.com/KevinOConnor/klipper/files/2849565/configBLTOUCH.txt)

CharlesStArnault, 2019-02-10:
I forgot to mention, but I have an mks 1.4 board

BlackStump, 2019-02-11:
try change your entry in stepper_z
from this
endstop_pin: z_virtual_endstop
to 
endstop_pin: probe:z_virtual_endstop

https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L49

closed 2019-02-18 by KevinOConnor