Klipper3d/klipper #1231: z over extrudion

sqrtmo, 2019-02-11:
Hi all,

yesterday saw that I have an issue with my z. It seems that my prints are higher than
expected. I did the math and set the "step_distance:" properly, so the head is moving as expected when move manually. But when I print a column I have deviation in z.

The column from the picture should be 40mm, but it si 42.2. Seems like it over extrudes in first ~10 mm and then prints correctly. 

I am using purse like printer with mesh bed levelling.

Attaching picture and log file.

Very enjoy using klipper!

Thanks in advance!

[image]
[klippy.log](https://github.com/KevinOConnor/klipper/files/2850007/klippy.log)

BlackStump, 2019-02-11:
did you read the bed_mesh example? in particular about fade_end
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L184
about scaling ?
https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L187

sqrtmo, 2019-02-11:
Thanks a lot!

Setting fade_end to 0 solves the problem, but what is the actual purpose of fade?

KevinOConnor, 2019-02-18:
I'm going to close this as it appears the original question was answered.  The purpose of "fade" (as I understand it) is to gradually remove the Z differences as the print height increases.  It's known to cause artefacts if there are significant Z variations in the mesh, however.  That's why it is not enabled by default.

-Kevin

closed 2019-02-18 by KevinOConnor